OctoPrint/OctoPrint #3171: Hide message from your printer box

guestisp, 2019-05-19:
Starting from the last update (1.3.11) i'm getting a lot of modal window "Message from your printer"

Some of them are good, but i'm also hitting some modal without any closing button, so the whole page became unusable. I have to manually remove the modal div from the Chrome inspector.

And, on top of that, as I didn't acknowledge the message, on page refresh Octoprint is still showing the same modal without the closing button

I've attached a screenshot. Where is the closing button ? Can I disable these invasive modals ?

[image]

foosel, 2019-05-20:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * steps to reproduce
  * your [exact version of OctoPrint](https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1)
  * your exact printer model & firmware version (ideally with a link to the sources)
  * the browser with which you are seeing this (underlying OS and full version please)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

For the record, this looks like for some reason your printer's firmware keeps triggering prompts on the connected host (here OctoPrint). To verify and also see *how* it triggers them I'd need the above mentioned `serial.log` however. The prompts are evaluated by the included Action Command Prompt Support plugin, if you disable that the prompts will stop showing up. However I'd prefer if we get to the bottom of why you are getting spammed by the firmware with these things since it might be an issue that needs to be fixed on both host and firmware side, so please provide the information as requested above. Thanks.

foosel, 2019-05-20:
Now look at that, I even added a config option for that:

[image]

guestisp, 2019-05-20:
Disabling the action command prompt plugin solved the issue.

I don't think it's the printer flooding the octoprint, because the message Is always shown even by reloading the webpage and not only on connection.

I don't know how it works internally but seems that there is a "last shown message" saves somewhere and if not acknowledge, it keep showing up. That's fine, as long there is a close button to acknowledge (and to close the modal)

foosel, 2019-05-20:
> I don't think it's the printer flooding the octoprint, because the message Is always shown even by reloading the webpage and not only on connection.

That's because prompts are stateful and the plugin "remembers" the last prompt the printer triggered (it should clear that up on disconnect however, so that might be a bug). 

> I don't know how it works internally but seems that there is a "last shown message" saves somewhere and if not acknowledge, it keep showing up. That's fine, as long there is a close button to acknowledge (and to close the modal)

Yeah... so... that's what should happen, but I cannot look into fixing this unless you provide me the logs I asked for.

guestisp, 2019-05-20:
> Yeah... so... that's what should happen, but I cannot look into fixing this unless you provide me the logs I asked for.

Busy working day today, i'll try to trigger this bug later this day

But, as developer, i think would be useful to see why some messages doesn't have at least an ok/close button. At least one button must always be shown and this would fix the issue

foosel, 2019-05-20:
> At least one button must always be shown and this would fix the issue

The prompt is *completely* defined by the firmware according to the protocol. So if the firmware doesn't define a button, there won't be a button. And I'd like to verify that this is the case here.

guestisp, 2019-05-20:
> The prompt is completely defined by the firmware according to the protocol. So if the firmware doesn't define a button, there won't be a button. And I'd like to verify that this is the case here.

Seems to be the case, but as you are forcing a modal window, without a button of any kind, the whole octoprint would be unusable. And as you are showing the same modal even after a refresh (because there wasn't an ack to the message), even subsequent access to octoprint would be unusable.

I understand that the firmware is defining which button to show, but a simple "close" button must be shown anyway. The firmware is not expecting a response of any kind, so is not asking for a particular button.

serial.log
```
2019-05-20 18:56:18,485 - Recv: ok N632 P0 B3
2019-05-20 18:56:18,491 - Send: N633 G1 X117.968 Y123.488 E32.14216*81
2019-05-20 18:56:18,524 - Recv: ok N633 P0 B3
2019-05-20 18:56:18,530 - Send: N634 G1 X117.968 Y122.977 E32.15752*94
2019-05-20 18:56:18,831 - Recv: ok N634 P0 B3
2019-05-20 18:56:18,837 - Send: N635 G1 X122.977 Y117.968 E32.37040*91
2019-05-20 18:56:18,869 - Recv: ok N635 P0 B3
2019-05-20 18:56:18,875 - Send: N636 G1 X122.466 Y117.968 E32.38576*90
2019-05-20 18:56:19,162 - Recv: ok N636 P0 B3
2019-05-20 18:56:19,169 - Send: N637 G1 X117.968 Y122.466 E32.57693*90
2019-05-20 18:56:19,200 - Recv: ok N637 P0 B3
2019-05-20 18:56:19,205 - Send: N638 G1 X117.968 Y121.955 E32.59229*80
2019-05-20 18:56:19,470 - Recv: //action:cancel
2019-05-20 18:56:19,471 - Cancelling on request of the printer...
2019-05-20 18:56:19,479 - Changing monitoring state from "Printing" to "Cancelling"
2019-05-20 18:56:19,507 - Send: M108
2019-05-20 18:56:19,627 - Recv: //action:prompt_end
2019-05-20 18:56:19,629 - Recv: //action:prompt_begin UI Abort
2019-05-20 18:56:19,630 - Recv: //action:prompt_show
2019-05-20 18:56:19,633 - Recv: ok N638 P0 B3
2019-05-20 18:56:20,452 - Recv:  T:211.04 /210.00 (71.75) B:59.95 /60.00 (857.25) @:74 B@:60
2019-05-20 18:56:20,503 - Recv: echo:enqueueing "M84 X Y Z E"
2019-05-20 18:56:20,521 - Recv: echo:enqueueing "G28XY"
2019-05-20 18:56:20,523 - Recv: ok P15 B1
2019-05-20 18:56:20,526 - Send: N639 M84*19
2019-05-20 18:56:22,453 - Recv:  T:210.73 /0.00 (72.12) B:59.83 /0.00 (857.81) @:0 B@:0
2019-05-20 18:56:22,549 - Recv: echo:busy: processing
2019-05-20 18:56:24,452 - Recv:  T:209.60 /0.00 (73.56) B:59.57 /0.00 (859.06) @:0 B@:0
2019-05-20 18:56:24,548 - Recv: echo:busy: processing
2019-05-20 18:56:25,461 - Recv: X:-5.00 Y:-5.00 Z:5.00 E:32.59 Count X:-400 Y:-400 Z:2000
2019-05-20 18:56:25,470 - Recv: ok N639 P15 B3
2019-05-20 18:56:25,473 - Send: N640 M104 T0 S0*35
2019-05-20 18:56:25,488 - Recv: ok N640 P15 B3
2019-05-20 18:56:25,493 - Send: N641 M140 S0*102
2019-05-20 18:56:25,501 - Recv: ok N641 P15 B3
2019-05-20 18:56:25,506 - Send: N642 M106 S0*103
2019-05-20 18:56:25,518 - Recv: ok N642 P15 B3
2019-05-20 18:56:25,521 - Changing monitoring state from "Cancelling" to "Operational"
2019-05-20 18:56:26,452 - Recv:  T:208.08 /0.00 (75.69) B:59.13 /0.00 (861.19) @:0 B@:0
2019-05-20 18:56:28,451 - Recv:  T:205.54 /0.00 (79.25) B:58.79 /0.00 (862.81) @:0 B@:0
2019-05-20 18:56:30,451 - Recv:  T:203.26 /0.00 (82.44) B:58.48 /0.00 (864.31) @:0 B@:0
2019-05-20 18:56:32,452 - Recv:  T:200.13 /0.00 (86.81) B:58.15 /0.00 (865.88) @:0 B@:0
```

octoprint.log
```
2019-05-20 18:51:04,928 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Cubo.gcode
2019-05-20 18:51:04,932 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=1200 --speed-y=1200 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Cubo.gcode
2019-05-20 18:51:06,300 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Cubo.gcode, owner: ale, user: ale
2019-05-20 18:51:06,316 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2019-05-20 18:51:06,323 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Cubo.gcode, owner: ale, user: ale
2019-05-20 18:51:06,335 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-05-20 18:51:06,346 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2019-05-20 18:51:06,667 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:51:08,474 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2019-05-20 18:51:34,797 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2019-05-20 18:51:36,695 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:52:06,748 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:52:36,799 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:52:47,166 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-05-20 18:53:06,852 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:53:36,905 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:54:06,959 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:54:37,015 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
[... cut ...]
2019-05-20 18:56:19,566 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Cubo.gcode, owner: ale, user: None
2019-05-20 18:56:21,041 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Cubo.gcode finished, needed 1.55s
2019-05-20 18:56:25,522 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2019-05-20 18:56:37,242 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-20 18:57:07,296 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
```

foosel, 2019-05-20:
> I understand that the firmware is defining which button to show, but a simple "close" button must be shown anyway. The firmware is not expecting a response of any kind, so is not asking for a particular button.

The thing is, if I allow the modal to be closed, I'm going against the agreed upon protocol here. The firmware is only supposed to send a modal like that *if it is blocked itself*. It should close it itself in such a case. Which it doesn't do here, it just defines a prompt, triggers it to show, then stops doing anything with it. That's *not* how prompts should be used.

But since apparently some firmware variant is already abusing this new feature and causing broken situations: I agree, to not make UX suffer here the only way is to go against the protocol and allow a manual close to handle situations like this.

guestisp, 2019-05-20:
So, what do you suggest firmware-side? I could try to issue a pull-request with a patch in Marlin

foosel, 2019-05-20:
It should define an abort button here. Or even get rid of this prompt altogether since I'm really unsure what it's even supposed to do tbh.

guestisp, 2019-05-20:
Do you have a reference to the protocol so i can evaluate a patch to Marlin?

foosel, 2019-05-20:
https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer and http://docs.octoprint.org/en/master/bundledplugins/action_command_prompt.html

In general I'd suggest to not abuse prompts for status messages but only for actual user interaction ("press button to continue", "select how to proceed").

We already have methods for status communication, we don't need to abuse something meant to be used to gather information from the user. I'm a nutshell, if it should only produce a display update on the built in controller instead of stopping and asking the user to press the button, the prompt actions are the wrong thing to use.

guestisp, 2019-05-20:
Based on the provided docs, I don't see anything forcing the firmware to set a button. This could lead to misinterpretation: as we (the firmware) don't ask for a button, the interface show a generic ok/cancel/ack/done button, just to not break the UX

Probably in this case the firmware doesn't expect any response back, that's why it doesn't ask for a specific button.

foosel, 2019-05-21:
If the firmware doesn't want a response back, it shouldn't be creating an *interactive prompt used for getting responses back from the user*. That is simply a misuse of the prompt actions leading to the exact spamming situation you found yourself in. If there's no need for any kind of action on part of the user there's no need to spam them with a prompt.

And the docs state that the firmware defines the buttons to be shown. So naturally if the firmware doesn't define buttons, no buttons are supposed to be shown 🤷

Side note: I'm very sure I understand this correctly since I defined this extension...

guestisp, 2019-05-21:
What I meant was: show this alert box, to notify the user that print was aborted via LCD screen "UI Abort" message. As firmware, i don't need to know that user as shown the alert, so I don't need an "OK" response sent back. Thus, there is no need (for the firmware) to add an OK button, it's up to UX to handle it's own UX properly without breaking.

It'a notification, not a prompt. Like the Android "toast" messages, just to make the idea.

Other the this, as developers, we all know that we can't totally rely on external software when developing an UX. We don't have control on external software, the UX should also handle (as much as possible) badly programmed external software.

foosel, 2019-05-21:
> As firmware, i don't need to know that user as shown the alert, so I don't need an "OK" response sent back.

Then you should not be using the *interactive prompts*. They are the wrong tool for this. We'd probably need to define a specific "please show a notification to the user" action command for that instead.

guestisp, 2019-05-21:
> Then you should not be using the interactive prompts. They are the wrong tool for this. We'd probably need to define a specific "please show a notification to the user" action command for that instead.

Yes, I think that would be the proper solution.

foosel, 2019-05-21:
So here's what I did on my end now. Prompts now can be dismissed with the close icon on the dialog. If so they show up as a little speech bubble in the navbar while they are still active on the firmware side. They also get removed altogether on disconnect (that not happening was a bug). 

Pushed to `maintenance`, soon `devel`, to be released as part of 1.3.12.

guestisp, 2019-05-21:
Can I cherry-pick this https://github.com/foosel/OctoPrint/commit/28229928436e89dc5034fd8807255f56bc6016a2 as quick workaround, before i launch the raspberry out of the window? It's very annoying

foosel, 2019-05-21:
Either that, or disable the plugin for now.

NovaViper, 2019-07-26:
Will this be added?? It's really frustrating that I can't hide the prompts

foosel, 2019-07-26:
Yeah, it's really frustrating that I constantly have to put workarounds into my project for misbehaving firmware indeed.

NovaViper, 2019-07-26:
Im in the middle of a 6 hour print and the interface is being blocked by the messages saying Resume SD card (I had a crash with the pi earlier in the very beginning of the print and had to reprint it from the SD card, hence why it's saying Resume SD), but I'm afraid that the PSU control plugin will cut off power to the printer as Im not directly printing from Octoprint, and to make things worse, I can't even tell that plugin to not do that because of the messages basically locking me out of Octoprint <_>

NovaViper, 2019-07-26:
*DANG IT* And of course the other plugins told it to stop heating, which made my print fail -.-

foosel, 2019-07-26:
I'm sorry that happened, but again, that is entirely caused by your printer firmware abusing the action command prompt specified for *interactive messages* for a blocking *non interactive* message, which is a firmware bug. Not OctoPrint. So there's no reason to become angry at me here or to vent your frustration on this bug tracker.

I've put in a workaround for this kind of misbehaviour, effectively hamstringing the whole feature, because *yet again* some firmware apparently can't behave. That workaround is part of the `maintenance` branch, what is going to be 1.3.12. Until that is released, which has a low priority at the moment due to development towards 1.4.0 that's been on the back burner for way too long thanks to maintenance including workarounds for broken firmware variants, there are alternative solutions to the problem listed above.

NovaViper, 2019-07-26:
I don't have the printer plugged in and it still keeps on displaying that prompt <_>Is there some sort of way to make it go away? Without having to unplug the pi? (I got a process running trying to compile PrusaSlicer onto it and it's taking a really long time)

foosel, 2019-07-26:
Go into the plugin manager, disable the plugin, then restart OctoPrint (*not* the Pi) from the system menu.

NovaViper, 2019-07-26:
But I literally can't click on anything because of the message.. even refreshing the page doesn't make a difference, the prompt just comes back after 5 seconds

foosel, 2019-07-26:
Then SSH into the machine, run `sudo service octoprint restart`. Should make the message go away. Then disable the plugin.

NovaViper, 2019-07-26:
I finally managed to make Octoprint restart! I think I may know how to fix the issue.. I was having issues with the resume action in the same way as the sd card resume action, and I recall using one of the Action Commands and made a little response (@resume) for Octoprint to respond to it. After this print, I'm going to go and experiment with it (and also send a message to the Marlin devs about the issue)

LMS0815, 2019-08-28:
> That's because prompts are stateful and the plugin "remembers" the last prompt the printer triggered (it should clear that up on disconnect however, so that might be a bug).

Any progress on this bug (OctoPrint Version 1.3.11)? 
If I reload the web-page (Chrome) I always get the last old message "UI Abort" - which is not valide any longer. - because I am printing.
[image]

Suggestions:
- Add a close button - I know ESC is closing it.
- Add some timeout and or validation against the current printer state.
- Get rid of it and integrate it to the _normal_ message system (Like other alerts)

NovaViper, 2019-09-02:
I've been looking at the logs and I see that Marlin only sends out the action *once*, it never actually continuously sends it. I second @LMS0815 's suggestion on adding a button to the prompts (even if the firmware doesn't actually define it)

SkullKill, 2019-09-03:
> I just got the Creality_DWINTest branch fork of the Marlin firmware and I can confirm the same issue as referenced in [InsanityAutomation/Marlin#57](https://github.com/InsanityAutomation/Marlin/issues/57)

the Creality_DWINTest branch is based of marlin 1.1.9

the one that i posted https://github.com/InsanityAutomation/Marlin/issues/57 is from the CrealityDwin_2.0 branch which is based off marlin 2.0 , so it looks like it is there for both marlin 2.0 and marlin 1.1.9

NovaViper, 2019-09-03:
@SkullKill Ah, I meant the 2.0 version of the branch. Sorry for the mix up!

foosel, 2019-09-04:
> So here's what I did on my end now. Prompts now can be dismissed with the close icon on the dialog. If so they show up as a little speech bubble in the navbar while they are still active on the [...]

commit 28229928 (2019-05-20): 🚸 action prompt: allow to cancel dialogs

Closes #3171
closed 2019-10-22 by foosel