OctoPrint/OctoPrint #3172: [Request] Option to power off USB ports when disconnected

amooz, 2019-05-19:
**Is your feature request related to a problem? Please describe.**
When my Creality printer is switched off and Octoprint is physically connected and powered on, there is still current flowing from the USB ports to the printer.  Searching the web this seems to be a common problem of Creality printers, maybe others as well. 

**Describe the solution you'd like**
When I click `Disconnect` in Octopi, it would power off all USB ports, maybe using the command `uhubctl -p 2 -a 0`.  When I click `Connect`, it first powers on the USB ports, `uhubctl -p 2 -a 1`.  A configuration option in Octoprint would be used to enable or disable this feature, and by default it would come disabled since not all printers suffer from this.  

This approach has the added (contextual) benefit of not powering USB accessories, such as USB cameras, while the printer is not in use.

**Describe alternatives you've considered**
The other options seem to be physical interrupts to the power coming from the RPI.  A USB isolator or 'cutting the red wire' in the USB cable. 

**Additional context**
This seems to affect Creality printers especially, and perhaps others as well.  Having a software solution should cover nearly all cases of this problem.  The community may even "run with this" and use it to control a relay which powers the printer up and down based on whether the USB ports are powered.

tedder, 2019-05-19:
[Here's a plugin designed to do that](https://community.octoprint.org/t/octoprint-usbcontrol/7659). You can see how it is problematic- there's no real provision for doing it in the Pi directly.

Some more:
* https://community.octoprint.org/t/usb-power-control-via-octoprint/1535
* https://community.octoprint.org/t/turn-off-usb-power-via-system-command-editor/7274

foosel, 2019-05-20:
OctoPrint is platform agnostic and there's no way to do that cross platform. As mentioned by @tedder it's not even possible to do it reliably when targeting only a specific platform. Closing as non feasible for core and due to an existing plugin that tries to achieve that goal.

closed 2019-05-20 by foosel