OctoPrint/OctoPrint #3173: Problem autodetect USB connection Ender 3

humbertosales, 2019-05-19:
#### What were you doing?

1. I turn on the printer Ender 3 and the raspberry pi (USB cable connected)
2. I waited till the server is online
3. I logged to Octoprint
4. I pushed the button "connect" with AUTO settings
5. I got a connection error and a red pop up appears with this message

**Error: Failed to autodetect serial port, please set it manually.**

I try turn on both without connected USB cable, but problem persist. Changing USB cable, problem persist. I try all combination, but nothing. :-(

#### What did you expect to happen?
Connect to the printer

#### What happened instead?
It appearde the message:
**Error: Failed to autodetect serial port, please set it manually.**

#### Did the same happen when running OctoPrint in safe mode?
I tried different configurations, but the error is aleatory. Intermittent!!
There are not plugin installed.
The Octopit is connected on Wifi 5 ghz.

#### Version of OctoPrint
OctoPrint version : 1.3.11

OctoPi version: 0.16.0

#### Operating System running OctoPrint

Client: Windows 10

Server: OctoPi version    : 0.16.0

#### Printer model & used firmware incl. version
Ender 3. Unique firmware.

#### Browser and version of browser, operating system running browser
Google Chrome latest

#### Link to octoprint.log
https://gist.github.com/humbertosales/1aa02de941946847160341b7804526ee

#### Link to contents of terminal tab or serial.log
https://gist.github.com/humbertosales/ce50f3ba490c43f4eddef2fbeebbf98c

#### Link to contents of Javascript console in the browser
packed_core.js?8c230bbf:15624 Starting dependency resolution...
packed_core.js?8c230bbf:15734 ... dependency resolution done
packed_core.js?8c230bbf:16016 Initial application setup done, connecting to server...
packed_core.js?8c230bbf:13728 Connected to the server
packed_core.js?8c230bbf:15994 Finalizing application startup
packed_core.js?8c230bbf:15875 Going to bind 32 view models...
packed_core.js?8c230bbf:15928 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?8c230bbf:15928 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?8c230bbf:15928 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?8c230bbf:2475 User op logged in
packed_core.js?8c230bbf:15967 ... binding done
packed_core.js?8c230bbf:15979 Application startup complete

#### Screenshot(s)/video(s) showing the problem:
[image]

foosel, 2019-05-20:
```
2019-05-19 15:59:57,040 - Serial port list: []
```

OctoPrint doesn't even see your printer, or there would be entries in the serial port list. Please check that it's properly plugged in and that the underlying OS detects it. Get in touch on the [community forum](https://community.octoprint.org) to get help with that.

closed 2019-05-20 by foosel

samuelriippa, 2019-09-26:
I got a same problem on ender 5. Here is how i solve it.  FIrst I boot raspberry. I have a usb cable connected to ender. Ender motherboard can have 5v power from raspberry. Then I open octoprint and connect to printer USB port must be seen on connection tab, if it says just auto and you cant see the port like /tty/USB0  reconnect the usb cabel in other raspberry port and refresh the octoprint page. Then when the connection is made I boot the ender PSU and start printing