OctoPrint/OctoPrint #3174: [Request] Make it easier to reload updated gcode

brendabell, 2019-05-19:
**Is your feature request related to a problem? Please describe.**

In previous versions, I was able to upload, load and print a model and immediately repeat that procedure with an updated version of the same model. With Octoprint 1.3.10, the load button is disabled. After changing my model, I have to load some other model first, then load the one I want to print.

In other words, the Load button is immediately grayed after a model is loaded. Users should be able to reload a model as often as they see fit.

**Describe the solution you'd like**

I'd like to see OctoPrint not disable the load button (unless printing is is progress).

foosel, 2019-05-20:
The load button is only disabled if the file is already loaded (or a print is in progress). Please check if you already had the file loaded (as in, it showed up in the State panel).

brendabell, 2019-05-21:
@foosel Maybe I'm misunderstanding how it's supposed to work. Yes I already have the file loaded. But the use case is I load the file and print. Then I either go back to Fusion 360 because I need to tweak the model *or* I go back to slic3r because I need to tweak the print settings. I upload a new version of the model. I want to print the updated version so I've been assuming it needs to be reloaded. The reason for that assumption is because the info displayed for the loaded model doesn't change to reflect the updated version.

If I don't load the model again, it's going to print the previously loaded version, yes?

ManuelMcLure, 2019-05-21:
I have found that Octoprint will automatically reload the file if you re-upload the file that is currently loaded. You don't need to press the reload button.

foosel, 2019-05-21:
What @ManuelMcLure said. No need to reload anything, if you upload a new version with the same name it's what will get used. I guess that needs to be made more clear in the UI though. Hm...

brendabell, 2019-05-21:
@foosel 👍

foosel, 2019-05-21:
[image]

1.3.12 will add the upload date to the information in the state panel, including seconds, so that it should be more clear that a reload has happened.

brendabell, 2019-05-21:
@foosel That's awesome! Thanks so much!

commit d40c3e62 (2019-05-21): 🚸 Add upload date to selected file information

Should make it more clear when an update due to a new upload happens.

Closes #3174
closed 2019-10-22 by foosel