OctoPrint/OctoPrint #3175: No upload of .gcode after update possible

Blackraz0r, 2019-05-20:
#### What were you doing?

1. Go to files and upload something or drag a gcode into the window
2. wait for the error: https://i.imgur.com/TJ6CNye.jpg
[image]
3. cant upload because of this

#### What did you expect to happen?
i expect the upload of .gcode files to happen.

#### What happened instead?
this error message:
https://i.imgur.com/TJ6CNye.jpg

#### Did the same happen when running OctoPrint in safe mode?
yes.

#### Version of OctoPrint
     OctoPrint 1.3.11 

#### Operating System running OctoPrint
     OctoPi 0.15.1 

#### Printer model & used firmware incl. version
Anycubic i3 Mega but not relevant thought

#### Browser and version of browser, operating system running browser
Chrome Version 74.0.3729.157 (Official Build) (64-bit)
Mozilla Firefox 66.0.5 (64-bit)

#### Link to octoprint.log
https://pastebin.com/SCLQZ5M3

#### Link to contents of terminal tab or serial.log
not applicatable

#### Link to contents of Javascript console in the browser
https://pastebin.com/L0GhgLK9

#### Screenshot(s)/video(s) showing the problem:
https://pastebin.com/SCLQZ5M3

I have read the FAQ.

Blackraz0r, 2019-05-20:
and yes it IS a valid gcode. all of the file i tried are. because the work on anonther pi with octoprint.

foosel, 2019-05-20:
I don't see the upload request arrive in that log. Can you share a screenshot of the network tab of the browser development tools after a reproduction with the failed POST to /api/files/local highlighted, and also tell me more about your network setup? Like, is there another reverse proxy between your browser and OctoPi?

foosel, 2019-05-20:
Also, make sure the files you are trying to upload are readable by *the browser*, in case you are running into #3133

Blackraz0r, 2019-05-20:
will try tomorrow again, have to sleep. thanks for the advice i will look into this =)

Blackraz0r, 2019-05-21:
Hey,

well i kind of found the error.

my pi with octoprint is connected via lan in a vlan in my dmz.
this totaly worked fine ( i also checked the firewall settings in my pfsense.

but what causes the errors is dual connectivity. i am also connected via wlan and and another vlan.
well the connection via the dmz doesnt work anymore. but the connection via wlan doas.

when i pull the cable and run on wlan only it works.
it also works if i disable wlan.

but having two connection prevents the upload. 

in know alot about routing and networking in general but nearly nothing about website programming. so im not shure what causes this type of error. for me it looks like octopi cant handle multiple connections from different ip's
networkside is everything ok. no packet loss, no blocking fw-rules or something like this.

so its somehow solved but somehow not.
it works well but there is no option for connection redundancy (wich i dont care for anyway)
i disabled wifi now and that works for me.

foosel, 2019-05-21:
OctoPrint can handle multiple connections from different IPs, but the session validation it uses might not like constant switching.

The above mentioned screenshots from the browser console's network tab might therefore still be interesting, it might give some insight what's happening. I'm not sure it's something I even could fix though, since I use a third party dependency for doing the whole session management stuff.

Still, could you tell me some more about your multi homed network setup? Are you accessing the Pi from different client IPs depending on the used NIC? Is it the same subnet or a different one? I'd like to at least understand what's going on here.

Blackraz0r, 2019-05-21:
well i can explain you my lab and how it works.

i have a pfsense on every location i administrate. also at my work ( where my printer is)
every location is connected via openvpn tunnel to every other location (fullmesh)
i tried to access octopy
from the workLAN (laptop) (vlan 2101 / 10.210.1.61)
to optopi lannic (vlan 2102 / 10.210.2.20) - no succes with uploads, can see website
and to wifi (vlan2101 / 10.210.1.20) - successful

well the second methot works so it somehow looks like it does because i am in the same subnet then.
but on the other hand its not really logic. my octoprint server has a default gateway. otherwise i wont be able to reach it via the dmz (vlan2102) also it wont be able to get updates vie the dmz. wich it does if i disconnect wifi.

i can supply another log from the browser tomorrow. sry fppr the late reply, reallife and stuf :/

foosel, 2019-05-27:
@Blackraz0r still can't make head nor tails of this issue, so if you could indeed provide the indicated logs, that would be lovely.

foosel, 2019-07-01:
@Blackraz0r ping? :)

closed 2019-07-25 by foosel