OctoPrint/OctoPrint #3178: Connection to printer failed on PrusaI3 MK2

Eriobis, 2019-05-22:
#### What were you doing?

1. Go to 'the octoprint main page
2. Click on connect
3. Wait few seconds
4. See error

#### What did you expect to happen?
I'd expect the printer to be connected
#### What happened instead?
The status is now disconnected, it's written to see the octoprint.log file
#### Did the same happen when running OctoPrint in safe mode?
No

#### Version of OctoPrint
1.3.11
Was working fine on 1.3.10

#### Operating System running OctoPrint
Raspbian
Linux octoprintmk2 4.4.23-v7+ #913 SMP Tue Oct 4 14:16:19 BST 2016 armv7l GNU/Linux

#### Printer model & used firmware incl. version
Prusa I3 Mk2 firmware 3.1.0

#### Browser and version of browser, operating system running browser
Unrelated but Chrome Version 74.0.3729.157 (Official Build) (64-bit)

#### Link to octoprint.log
https://pastebin.com/QbcnpnuM

#### Link to contents of terminal tab or serial.log
https://pastebin.com/KeyJ0aPa

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2019-05-23:
**Two** optional plugins are failing:
```
2019-05-22 23:48:24,844 - octoprint.server.views - ERROR - Error while retrieving template data for plugin navbartemp, ignoring it
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.11-py2.7.egg/octoprint/server/views.py", line 657, in fetch_template_data
    configs = implementation.get_template_configs()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_navbartemp/__init__.py", line 82, in get_template_configs
    if self.sbc.is_supported:
AttributeError: 'NoneType' object has no attribute 'is_supported'
```

and:
```
2019-05-22 23:48:29,890 - octoprint.plugin - ERROR - Error while calling plugin stats
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.11-py2.7.egg/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats/__init__.py", line 515, in on_event
    self.statDB.execute(sql)
AttributeError: 'StatsPlugin' object has no attribute 'statDB'
```

One of those, presumably, is crashing at connection time:
```
TypeError: on_comm_temperature_update() takes exactly 3 arguments (4 given)
```

There's an error that follows with `notify/pnotify.maxheight.js`, but given the errors before that it's hard to know if it's related. Disable those plugins and share another log if it continues.

foosel, 2019-05-23:
> #### Did the same happen when running OctoPrint in safe mode?
> 
> No

Not a bug in OctoPrint then but one of your installed third party plugins. I suggest to try what @tedder suggested. Closing this though since it's nothing in core OctoPrint (otherwise it would still fail in safemode). Comments are still open however.

Eriobis, 2019-05-23:
My bad, I haven't tried it in safe mode. Will do when i get a chance. I saw an error about the navbar getting 4 args instead of 3 and I saw that the parsing regex has changed between version. Could this be a clue to the reason of the fail ?

foosel, 2019-05-23:
Why did you answer "No" then to the question if you could reproduce the issue in safemode that also made it very clear to *try to do that*? ;)

Eriobis, 2019-05-23:
> Why did you answer "No" then to the question if you could reproduce the issue in safemode that also made it very clear to _try to do that_? ;)

> If you can't test this in safe mode, state why.
I had a crying baby behind me and didn't have time to properly fill the issue :)
I didn't want to forget to fill an issue in case someone else may be affected by this kind of bug. I also assumed the prusa mk2 is quite popular and if there was a compatibility issue with the FW it would have been flagged in the previous RC. Let me double check the plugins tonight and I will close the bug if it can't be reproduced in safe mode.

hermy65, 2019-05-23:
@foosel @tedder Im having the same issue with my MK3 both in regular AND in Safe Mode after updating to 1.3.11.   

While looking through my log i did see that two plugins had errors (cancelobject and canvas) which i disabled but still was unable to connect in regular mode. Put it into Safe Mode and im still unable to get it to connect to my MK3.

Log: https://pastebin.com/SQriuQuM

Terminal: https://pastebin.com/d17EKmPi

tedder, 2019-05-23:
Hermy65, add logs with it running in safe mode.

foosel, 2019-05-23:
@hermy65 auto detection can be very hit and miss (so much so that I've actually thought about removing it). Try manually selecting one of the two USB serial devices that your OS has found and see if it works that way. My own MK3 identifies as `/dev/ttyACM0` and connects just fine.

hermy65, 2019-05-23:
@foosel strange, never had an issue with auto before, took a couple tries but it finally connected on /dev/ttyACM0 - thanks!

Eriobis, 2019-05-24:
Update : Here's the log while running it in safe mode.
[https://pastebin.com/GqWMtWwF](https://pastebin.com/GqWMtWwF)

The first error shown is : 
> 2019-05-24 21:43:02,335 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.11-py2.7.egg/octoprint/util/comm.py", line 1816, in _monitor
    self._callback.on_comm_temperature_update(self.last_temperature.tools, self.last_temperature.bed, self.last_temperature.chamber)
TypeError: on_comm_temperature_update() takes exactly 3 arguments (4 given)

Eriobis, 2019-05-26:
I got it working by reinstalling octoprint from scratch.
I guess that the setup.py clean and install doesn't well work anymore when updating the git repo.

closed 2019-05-26 by Eriobis

foosel, 2019-05-27:
> I guess that the setup.py clean and install doesn't well work anymore when updating the git repo.

That's odd, it works just as it should over here (I switch constantly between branches and minor versions).

Eriobis, 2019-05-27:
> > I guess that the setup.py clean and install doesn't well work anymore when updating the git repo.
> 
> That's odd, it works just as it should over here (I switch constantly between branches and minor versions).

Could it be that the virtual environment had old dependencies/plugins and they weren't updated when doing setup.py install ?

My method was : 
`git fetch
git checkout master
./venv/bin/python setup.py clean
./venv/bin/python setup.py install
Then restart the octoprint server`

foosel, 2019-05-28:
That sounds like it's missing a step actually... `git fetch` only updates the local mirrors of the remote so to speak, it doesn't actually update your local branch.

 Also these days I rather recommend `pip install .` instead of `python setup.py install`.

Here's what I'd do:

```
git checkout master
git pull
./venv/bin/python setup.py clean
./venv/bin/pip install .
```