OctoPrint/OctoPrint #3182: AttributeError: 'module' object has no attribute 'cpu_freq'

hpoleselo, 2019-05-27:
Hello, i'm trying to run Octoprint on Ubuntu 16.04 and installed supposedly correctly, i.e using Python 2.7.
After following all the steps to install Octoprint on my machine, i tried to run Octoprint with the following command:
`octoprint serve`

And the result is:
`/usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.3) or chardet (2.3.0) doesn't match a supported version!
  RequestsDependencyWarning)
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
2019-05-27 10:53:34,139 - octoprint.startup - INFO - ******************************************************************************
2019-05-27 10:53:34,140 - octoprint.startup - INFO - Starting OctoPrint 1.3.11
2019-05-27 10:53:34,140 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2019-05-27 10:53:34,140 - octoprint.startup - INFO - Reason for safe mode: problem during last startup
2019-05-27 10:53:34,141 - octoprint.startup - INFO - ******************************************************************************
2019-05-27 10:53:34,150 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/octoprint/plugins, /home/fastenufba/.octoprint/plugins and installed plugin packages...
2019-05-27 10:53:34,276 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-05-27 10:53:34,411 - octoprint.plugins.octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2019-05-27 10:53:34,412 - octoprint.plugin.core - INFO - Found 14 plugin(s) providing 14 mixin implementations, 22 hook handlers
2019-05-27 10:53:34,422 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-05-27 10:53:34,425 - octoprint.server - INFO - Intermediary server started
2019-05-27 10:53:34,425 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/octoprint/plugins, /home/fastenufba/.octoprint/plugins and installed plugin packages...
2019-05-27 10:53:34,427 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2019-05-27 10:53:34,486 - octoprint.plugin.core - INFO - Found 14 plugin(s) providing 14 mixin implementations, 22 hook handlers
2019-05-27 10:53:34,489 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/fastenufba/.octoprint/uploads...
2019-05-27 10:53:34,490 - octoprint.filemanager.storage - INFO - ... file metadata for /home/fastenufba/.octoprint/uploads initialized successfully.
2019-05-27 10:53:34,891 - octoprint.util.pip - INFO - Using "/usr/bin/python2.7 -m pip" as command to invoke pip
2019-05-27 10:53:35,304 - octoprint.util.pip - INFO - Version of pip is 19.1.1
2019-05-27 10:53:35,304 - octoprint.util.pip - INFO - pip installs to /usr/lib/python2.7/dist-packages (writable -> no), --user flag needed -> yes, virtual env -> no
2019-05-27 10:53:35,304 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-05-27 10:53:35,305 - octoprint.plugin.core - INFO - Initialized 14 plugin implementation(s)
2019-05-27 10:53:35,307 - octoprint.plugin.core - INFO - 14 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/errortracking
|  Force Login (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/logging
|  Plugin Manager (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /usr/local/lib/python2.7/dist-packages/octoprint/plugins/virtual_printer
2019-05-27 10:53:35,308 - octoprint.environment - ERROR - Error while detecting hardware environment
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/octoprint/environment.py", line 105, in _detect_hardware
    cpu_freq = psutil.cpu_freq()
AttributeError: 'module' object has no attribute 'cpu_freq'
2019-05-27 10:53:35,309 - octoprint.environment - INFO - Detected environment is Python 2.7.12 under Linux (linux2). Details:
|  hardware:
|    cores: unknown
|    freq: unknown
|    ram: unknown
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 19.1.1
|    version: 2.7.12
2019-05-27 10:53:35,311 - octoprint.server - INFO - Reset webasset folder /home/fastenufba/.octoprint/generated/webassets...
2019-05-27 10:53:35,311 - octoprint.server - INFO - Reset webasset folder /home/fastenufba/.octoprint/generated/.webassets-cache...
2019-05-27 10:53:35,401 - octoprint.server - INFO - Shutting down intermediary server...
2019-05-27 10:53:35,426 - octoprint.server - INFO - Intermediary server shut down
2019-05-27 10:53:35,427 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/usr/local/bin/octoprint", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/octoprint/__init__.py", line 609, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/cli/server.py", line 206, in serve_command
    ignore_blacklist)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/cli/server.py", line 118, in run_server
    octoprint_server.run()
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/__init__.py", line 697, in run
    self._server = util.tornado.CustomHTTPServer(self._tornado_app, **server_kwargs)
  File "/usr/lib/python2.7/dist-packages/tornado/util.py", line 222, in __new__
    instance.initialize(*args, **init_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/octoprint/server/util/tornado.py", line 744, in initialize
    tornado.httpserver.HTTPServer.initialize(self, *args, **kwargs)
TypeError: initialize() got an unexpected keyword argument 'trusted_downstream`

tedder, 2019-05-27:
It's hard to read the log because it's all jammed together, rather than each line starting properly. Can you reformat?

hpoleselo, 2019-05-27:
So i realized there was an older version from tornado (4.2.2) and it was somehow implicitly conflicting with the one which was trying to be installed... 
So i just:
```
sudo rm -rf /usr/local/lib/python2.7/dist-packages/tornado
sudo rm -rf /usr/lib/python2.7/dist-packages/tornado
```
And then:
`sudo pip2.7 install . --ignore-installed ${tornado}`

foosel, 2019-05-27:
And why did you decide to not follow [the official install instructions](https://github.com/foosel/OctoPrint/#installation), which explicitly tell you to install into a *virtual environment* instead of doing a global install, to avoid version conflicts like this among other things? ;)

closed 2019-05-27 by foosel

hpoleselo, 2019-05-27:
In my case i'm trying to integrate OctoPrint with ROS and creating the virtual environment was not possible, but thanks for the heads up!