OctoPrint/OctoPrint #3186: GCODE viewer stopped working

jarjartee, 2019-05-31:
#### Recently did a FRESH install of Octoprint - all features working up until 2 days ago when gcode viewer is NOW NOT showing anything but a blank grid ??

1. ...Installed fresh Octoprint on RaspberryPi 3B+
2. ...Installed Plugins 
3. ...

#### For GCODEViewer to show codes ??

#### Blank Grid showing

#### YES

#### 1.3.11

#### MacOS 10.13.6

#### Ender 3 / Marlin 1.19 bugfix

#### Safari 12.1.1 / MacOS 10.13.6

#### [octoprint.log.2019-05-29.txt](https://github.com/foosel/OctoPrint/files/3241460/octoprint.log.2019-05-29.txt)
[plugin_pluginmanager.log.2019-05-29.txt](https://github.com/foosel/OctoPrint/files/3241461/plugin_pluginmanager.log.2019-05-29.txt)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-05-31:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * *exact* steps to reproduce, ideally with step by steps descriptions of what buttons to click where and files provided to test against
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)
  * screenshots or maybe a video showing the problem
  * since this looks like an issue related to specific GCODE file properties, one or more GCODE files with which to reproduce this

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

foosel, 2019-06-03:
@jarjartee Ping? :) I really need some more info from you here, otherwise this cannot be looked into and will be closed.

jarjartee, 2019-06-03:
Sorry - had a bit of a nightmare with crashes and reconfiguring firmwares so started fresh - new install - now works - shows 'graphic paths' gcode if that makes sense(yes I'm a n00b) - closing

closed 2019-06-03 by jarjartee