OctoPrint/OctoPrint #3187: [Request]Prompt user for action when printer reports "echo:busy: paused for user" or similar messages

Patronics, 2019-06-01:
**Is your feature request related to a problem? Please describe.**
It's frustrating when I start a routine print in Octoprint, only to check back later expecting a finished print and find that the print actually never started (or stopped partway through), as the printer was waiting for me to interact, such as selecting which filament to use when the print starts, or to reload filament.

**Describe the solution you'd like**
It would be useful if Octoprint provided a pop-up alert when the printer is waiting for interaction, and reporting that it's paused and waiting for user interaction.

**Describe alternatives you've considered**
A partial implementation could potentially be implemented by the [M117PopUp plugin](https://plugins.octoprint.org/plugins/M117PopUp/), but only in situations where the gcode being sent from the printer could be edited to include it. So it could probably report the initial filament selection request, but not if filament ran out later or other issues.

**Additional context**
The feature should probably also be exposed in some API, so plugins like "Discord Remote" can implement notifying users remotely as well.

reloxx13, 2019-06-01:
did u check on action commands?
```

/**
 * Host Action Commands
 *
 * Define host streamer action commands in compliance with the standard.
 *
 * See /**
 * Host Action Commands
 *
 * Define host streamer action commands in compliance with the standard.
 *
 * See https://reprap.org/wiki/G-code#Action_commands
 * Common commands ........ poweroff, pause, paused, resume, resumed, cancel
 * G29_RETRY_AND_RECOVER .. probe_rewipe, probe_failed
 *
 * Some features add reason codes to extend these commands.
 *
 * Host Prompt Support enables Marlin to use the host for user prompts so
 * filament runout and other processes can be managed from the host side.
 */
//#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PROMPT_SUPPORT
#endif
 * Common commands ........ poweroff, pause, paused, resume, resumed, cancel
 * G29_RETRY_AND_RECOVER .. probe_rewipe, probe_failed
 *
 * Some features add reason codes to extend these commands.
 *
 * Host Prompt Support enables Marlin to use the host for user prompts so
 * filament runout and other processes can be managed from the host side.
 */
//#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PROMPT_SUPPORT
#endif
```

https://reprap.org/wiki/G-code#Action_commands

octoprint cannot do anything if the fw does not tell it to octoprint. 
action commands does this in the fw.

Patronics, 2019-06-01:
Thanks for the suggestion, but I don't think action commands is the best solution, as that would require a modified firmware installed on each printer, rather than a relatively simple modification to Octoprint. And Octoprint clearly is already processing the "echo: busy: xxxxxxx"  messages, as it reports when it first receives one that the printer has been identified as supporting the busy protocol, and adjusting timeout accordingly.

Patronics, 2019-06-01:
Upon further investigation, I think this is pretty simple to implement as a plugin, I'm going to try to modify m117PopUp plugin to see if I can get it working. This'll be my first time writing an Octoprint plugin though, so no guarantee I'll be successful or done in a timely fashion. If anyone else with more experience with plugin development wants to try as well, they're certainly welcome to.

Patronics, 2019-06-02:
I have successfully implemented this as a plugin, which is available here: [https://github.com/Patronics/OctoPrint-PrinterAlerts](https://github.com/Patronics/OctoPrint-PrinterAlerts). I will try to clean up the remnants from M117PopUp over the next few days, then submit it to the official Plugin Repository 😄

closed 2019-06-02 by Patronics

foosel, 2019-06-03:
@Patronics That's the spirit! 😊👍