OctoPrint/OctoPrint #3188: WebSocketClosedError

pusewicz, 2019-06-01:
#### What were you doing?

I'm not entirely sure what triggered this, as I was testing OctoPod and OctoClient iOS apps, but this is what happened in the logs:

```
2019-06-01 21:25:35,080 - tornado.application - ERROR - Exception in callback <functools.partial object at 0xb209c1e0>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/ioloop.py", line 605, in _run_callback
    ret = callback()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/transports/websocket.py", line 94, in <lambda>
    self.server.io_loop.add_callback(lambda: self.send_pack(message, binary))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/sockjs/tornado/transports/websocket.py", line 89, in send_pack
    self.write_message(message, binary)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/websocket.py", line 249, in write_message
    raise WebSocketClosedError()
WebSocketClosedError
```

#### Version of OctoPrint

OctoPrint 1.3.11

#### Operating System running OctoPrint

OctoPi 0.16.0

#### Printer model & used firmware incl. version

Creality Ender-3 Pro

#### Browser and version of browser, operating system running browser

Chrome

foosel, 2019-06-03:
As far as I read that it is harmless, caused by a sudden disconnect of a connected client, handled and only logged for reasons of transparency. At least this is not the first time that I've seen it, usually when suddenly killing browsers. I might actually look into suppressing this kind of messages.

But thanks for reporting anyhow!

foosel, 2019-06-03:
Oh, that has actually already been taken care of on `maintenance` and `devel` in preparation for 1.3.12 in commit 7dae700215c50d04c7d2b4c07801048c5d96f250

pusewicz, 2019-06-03:
Wunderbar! Viel danke!

closed 2019-06-03 by pusewicz