OctoPrint/OctoPrint #3189: Thermal Runaway Warning when attempting to Pause and Resume Prints

milmber, 2019-06-03:
#### What were you doing?

1. I configured the following GCode to be executed when pausing and resuming a print

After print job is paused:
```{% if pause_position.x is not none %}
; relative XYZE
G91
M83

; retract filament of 0.8 mm up, move Z slightly upwards and 
G1 Z+5 E-0.8 F4500

; absolute XYZE
M82
G90

; move to a safe rest position, adjust as necessary
G1 X0 Y0
{% endif %}
```

Before print job is resumed:
```{% if pause_position.x is not none %}
; relative extruder
M83

; prime nozzle
G1 E-0.8 F4500
G1 E0.8 F4500
G1 E0.8 F4500

; absolute E
M82

; absolute XYZ
G90

; reset E
G92 E{{ pause_position.e }}

; WARNING!!! - use M83 or M82(exruder absolute mode) according what your slicer generates
M82 ; extruder relative mode

; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500

; reset to feed rate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}
```
2. Printed the [3D Benchy Model](https://www.thingiverse.com/thing:763622).
3. The print was initiated from Cura 4.1 to an OctiPi instance running on a Raspberry Pi 3B+
4. The target bed temperature was 65C and the target hot-end temperature was 200C.
5. After 4 layers of successful printing with no issues I paused the print using the Pause print function in OctoPrint via the WebUI. 

#### What did you expect to happen?

1. The print to be paused when pressing the "Pause" button.
2. The print head to be moved out of the way from the print.
3. The temperatures for the hot-end and bed to drop.
4. Shut down the printer but keep Octoprint running.
5. Power on the printer
6. Resume the print from Octoprint using the "Resume" button.

#### What happened instead?

1. The print was paused.
2. The print head moved out of the way from the print.
3. The temperatures of the hot-end and the bed started to drop as expected
[... 79 lines cut ...]
2019-06-02 19:00:05,701 - tornado.access - WARNING - 409 GET /api/printer (fe80::85f:f80c:ba7d:710b) 22.37ms
2019-06-02 19:00:07,961 - tornado.access - WARNING - 400 POST /api/connection (fe80::85f:f80c:ba7d:710b) 57.77ms
2019-06-02 19:00:16,631 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-02 19:00:16,843 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2019-06-02 19:00:16,851 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2019-06-02 19:00:16,886 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-06-02 19:00:17,086 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-06-02 19:00:18,836 - tornado.access - WARNING - 409 GET /api/printer (fe80::85f:f80c:ba7d:710b) 19.61ms
2019-06-02 19:00:20,223 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2019-06-02 19:00:20,235 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-06-02 19:00:20,298 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Creality 3D"
2019-06-02 19:00:20,301 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
```

#### Link to contents of terminal tab or serial.log

Unable to provide as serial.log was not enabled and the Raspberry Pi became unresponsive.

#### Link to contents of Javascript console in the browser

N/A - this was not a UI issue.

#### Screenshot(s)/video(s) showing the problem:

None

foosel, 2019-06-03:
Ah thermal runaway is *always* triggered by the firmware and entirely outside of the control of OctoPrint. You'll need to figure out why your GCODE causes your firmware to trigger a thermal runaway (if the GCODE is even the cause here). This is not something that happened on the RPi (even though it's temperature going up as well sounds like you also have some possible cooling issues that you should look into) and everything to do with your firmware detecting some kind of abnormal sensor behaviour that made it think the hotend was stuck at on or the thermistor had fallen out of the hotend.

closed 2019-06-03 by foosel

dman61, 2019-07-10:
I had the same issue. Depending on your configuration you must change the following line in the print job resume script:

; WARNING!!! - use M83 or M82(exruder absolute mode) according what your slicer generates
M82 ; extruder relative mode

On mine Cura outputs a M83. If you resume using the default M82 then it will cause your printer to potentially trigger a thermal runaway. Regardless the print will not resume correctly. You must change this line to M83. If you want to verify how your slicer sets up your printer then open your GCODE file in a text editor. Within the first couple of lines you will either see a M82 or M83. You should use the same in the resume script.