OctoPrint/OctoPrint #3193: Websocket returns 403 for third party clients

foosel, 2019-06-05:
### Problem

As reported in there appears to be an issue with third party clients such as OctoPod connecting to the web socket if it's located behind a reverse proxy, causing an HTTP 403 to be thrown. 

Further investigation needed to figure out if this is a buggy implementation in the websocket endpoint or a problem with the calling client code.

Pinging @gdombiak and @evanquinnalaska.

foosel, 2019-06-05:
@gdombiak What exactly are you sending as `Host` and `Origin` headers?

foosel, 2019-06-05:
Based on the [implementation](https://github.com/tornadoweb/tornado/blob/8e9e75502ff910629663c4cdd7779d43ea2dd150/tornado/websocket.py#L175-L192), just not setting an `Origin` header at all should make `check_origin` not to even be called and hence not cause any issues 🤔

That goes against [this comment](https://github.com/foosel/OctoPrint/issues/1881#issuecomment-498737832) though:

> after OctoPrint 1.3.11 was released I had to push a new version of OctoPod to include the Origin header since it was giving a 403 error under certain circumstances

@gdombiak can you elaborate on these "certain circumstances"?

gdombiak, 2019-06-06:
Hi @foosel, I got confused on my initial answer. OctoPod is setting the _Host_ header but not the _Origin_ header. 

As for the _Host_ header I'm parsing the URL that was provided to connect to OctoPrint and use the hostname (port not included) as the header value.

I did a quick test the other day and reverted the fix that added the _Host_ header to see when it was failing. It was failing when I was connecting via the LAN (no proxy or anything in between the app and OctoPrint). It was working fine when going through my reverse proxy and the reason is that nginx is configured with something like `proxy_set_header Host            $host;`

Let me know if you need any other info.
Thanks,
Gaston

gdombiak, 2019-06-06:
Question for @evanquinnalaska, are you using a reverse proxy to connect to OctoPrint? I wonder if this error could be produced if your reverse proxy is not forwarding the _Host_ header. It wouldn't matter that OctoPod or any client is passing it if the proxy (or anything in between) is dropping it.

Just an idea to check out.
Gaston

foosel, 2019-06-07:
The port definitely needs to be part of the `Host` header: https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Host

foosel, 2019-06-07:
@gdombiak Would it be possible for you to provide me with some logged request/response pairs of your observations? I'd love to see exactly what goes over the line, and what comes back in case of failure.

gdombiak, 2019-06-07:
@foosel I think that you nailed it (at least one way to reproduce this problem). 

I changed the port to not be the _standard_ one and I now get a a 403 when not including port number in _Host_ header. Entry that appears in the log:
```
2019-06-07 20:33:17,318 - tornado.access - WARNING - 403 GET /sockjs/websocket (98.232.252.169) 5.01ms
```

I added the port number to the _Host_ header and the 403 error is now gone.

Let me know if you still need request/response logs for more analysis. I filed https://github.com/gdombiak/OctoPod/issues/189 and fixed it for upcoming release. Great catch. Thanks for helping fix this.

Gaston

foosel, 2019-06-11:
@gdombiak I'd still be interested in a request/response log because I'd like to understand why the port number mismatch causes a 403 to be returned, and right now I'm failing at that 🤔 Also not closing this until I've figured this out.

foosel, 2019-06-11:
I took a look at the websocket library you use and found this:

https://github.com/daltoniam/Starscream/blob/a2ed45c0b2f996cb8c335c4f270ecc68c3bd4c0f/Sources/Starscream/WebSocket.swift#L469-L477

As far as I understand these lines, it will *force* an `Origin` header to be set on the request that matches the domain + port combo as found in the request URL. E.g. "octopi.local:1234" if the request URL is something like "http://octopi.local:1234/". If that now gets combined with setting an incomplete `Host` header as you so far have been doing (`octopi.local` instead of `octopi.local:1234`) this will in turn cause the origin check in Tornado's websocket implementation to fail:

https://github.com/tornadoweb/tornado/blob/8e9e75502ff910629663c4cdd7779d43ea2dd150/tornado/websocket.py#L358-L412

Enabling CORS support circumvents this check/gives it an alternative outcome in the overwritten version of this method, but if CORS is disabled the default behaviour applies, which is to deny access:

https://github.com/foosel/OctoPrint/blob/c90f556fe92fb37693fc2a3ecf67e74ed01dd12f/src/octoprint/vendor/sockjs/tornado/websocket.py#L18-L34

So this behaviour looks like it's Starscream specific since for some reason Starscream always enforces an `Origin` header. I'm not entirely sure *why* it does that, from my understanding that is not a requirement for a standalone websocket client connection and a quick look into the RFC didn't unearth something either. Then again I have *no* idea at all of Swift. 

In any case, mystery solved I think - I'm pretty sure a request log from your app would show the presence of an `Origin` header even though you are not actively setting it, which makes the whole observed behaviour completely explainable. Still would like to check, cautiously closing this for now though.

closed 2019-06-11 by foosel

foosel, 2019-06-11:
Just for reference, quote from [RFC6455](https://tools.ietf.org/html/rfc6455), emphasis mine:

>    The |Origin| header field [RFC6454] is used to protect against
>    unauthorized cross-origin use of a WebSocket server by scripts using
>    the WebSocket API in a web browser.  The server is informed of the
>    script origin generating the WebSocket connection request.  If the
>    server does not wish to accept connections from this origin, it can
>    choose to reject the connection by sending an appropriate HTTP error
>    code.  **This header field is sent by browser clients; for non-browser
>    clients, this header field may be sent if it makes sense in the
>    context of those clients.**

So definitely a can, not a must. And not a bug to send it either, but if it's sent it certainly requires a correct `Host` header.