OctoPrint/OctoPrint #3198: Start GCode sequence set in Octoprint doesn't seem to execute sequentially

drewzh, 2019-06-18:
Octoprint version 1.3.11

When taking the exact same GCode start sequence out my slicer and moving it into Octoprint. The gcode steps seems to be execute all at once without taking into account the wait for heater steps to complete before moving onto the homing/abl etc.

The resulting behaviour is that the nozzle purge and subsequent steps are attempted with a cold nozzle.

I actually prefer the start sequence to be handled by the host, to allow me to change the various sequences without having to produce new gcode - I also think it makes more sense this way in general.

Is there certain limitations to the start gcode in octoprint that I'm unaware of? Should it not be used for things like homing and ABL?

Here's my start gcode, please note that moving this directly from Octoprint into my slicer fixes the problem immediately and the sequence then waits for the bed to heat up before moving onto the rest of the print (note I'm aware that the nozzle doesn't wait in this code).

; Ender 3 Custom Start G-code
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G90 ; Set absolute
G28 ; Home all axes
G29 ; ABL
M85 S600; Shutdown after 10 minute idle
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X2 Y20 Z0.3 F5000.0 ; Move to start position
G1 X2 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

foosel, 2019-06-19:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * a full [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1) showing the issue

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

foosel, 2019-07-01:
Hey there, there is sadly still some information missing. Could you provide it please? Otherwise this ticket will have to be closed 😕

drewzh, 2019-07-01:
@foosel Logs attached. So current behaviour with gcode added to octoprint and removed from .gcode file is all start gcode is ran before any heaters are activated. Only the print is started when the heaters are ready, thus start purges etc are missed.

https://gist.github.com/drewzh/f2d63efada1cc1417bce024a3e1006ee
https://gist.github.com/drewzh/0604e0a36b03625e583584f4d0f2ecc7

foosel, 2019-07-01:
Is that `serial.log` actually showing the reported issue? Because what I'm seeing is an `M109` being issued and the firmware entering a blocking heatup just as expected.

I took another look at your initial report and noted that your temperature commands contained temperature placeholders - that's not supported in those scripts and also not sanitized away by OctoPrint, so if you left that stuff in, I'm not surprised these lines where just rushed through by the firmware.

In general consider the GCODE scripts more of a tool to do *OctoPrint specific* stuff before every print that is unrelated to your particular model you are going to print. Personally I'd view things like a leveling routine, temperature settings and such as something that is an inherent part of the print job and hence should be part of the sliced GCODE file, *not* the host's additional scripts.

drewzh, 2019-07-01:
@foosel The 2 'wait for X heater' commands are a little misleading - I neglected to remove them when I moved the gcode over to Octoprint (they're Cura specific variables). Removing them has no effect and that's how I produced the last 2 logs, with those 'waits' removed.

The sequence starts at line 250. G28 is on 257, G29 is on 288. There's the purging that occurs from 372 onwards (resulting in cold purge). The print finally starts after the heaters are ready, but due to the start sequence already being processed by Octoprint, the nozzle purges itself as part of the skirt.

Ah, so this feature isn't intended for actual traditional 'start gcode' sequences like homing, bed leveling and purging nozzle etc? Kinda begs the question of what IS IT useful for other than those sorts of things (forgive my ignorance). If this truly isn't the place for providing a global start gcode and gcode must be 'baked' into the print files, maybe a warning should be added to the UI to indicate it's actual purpose?

I found the concept of separating out the printer specific routines from the actual model data to be quite appealing. It allows me to change the printer specific gcode without having to reproduce all the gcode again.

foosel, 2019-07-02:
> The sequence starts at line 250. G28 is on 257, G29 is on 288. There's the purging that occurs from 372 onwards (resulting in cold purge). The print finally starts after the heaters are ready, but [...]

And that is exactly how you've programmed things to be here. Your start script doesn't contain heat up commands, so the purge can't proceed. Your sliced file *does* contain stuff, so the print will proceed from then out. 

I have a feeling there's a bit of a misunderstanding here about the order that things are processed, so here's a breakdown of what happens (assuming a start - pause - resume - finish run):

  # Print job is started
  # State switches to "Starting"
  # The "beforePrintStarted" GCODE Script is sent to the printer line by line
  # State switches to "Printing"
  # The printed GCODE file is sent to the printer line by line
  # Print job is paused
  # State switches to "Pausing"
  # Sending of the GCODE file is paused
  # The "afterPrintPaused" GCODE Script is sent to the printer line by line
  # State switches to "Paused"
  # Print job is resumed
  # State switches to "Resuming"
  # The "beforePrintResumed" GCODE Script is sent to the printer line by line
  # State switches to "Printing"
  # Sending of the GCODE file is resumed
  # The end of the GCODE file is reached
  # State switches to "Finishing"
  # The "afterPrintDone" GCODE Script is sent to the printer line by line
  # State switches to "Operational"

Note that no parts of the printed file are magically injected into the scripts. If you want to do stuff that requires a heated up extruder in the "beforePrintStarted" script, you need to include that there. E.g. if you want to purge, you need to heat up. There's no reliable way for OctoPrint to separate the prescripts of existing GCODE files from the actual model printing, so there's no way to inject some kind of user defined script in between that and the actual printing. 

Also note that it says that the "beforePrintStarted" script is run "before print job starts", not "while it starts" or something like that.

> Kinda begs the question of what IS IT useful for other than those sorts of things (forgive my ignorance).

To be honest, I included it mostly to mirror the "afterPrintCancelled" script that I included (which is used to shut off heaters and maybe move the bed to some meaningful place). People use it for `M117` shenanigans as far as I know. Note that you *can* of course use it for separating your standard slicer "preamble" from your files if you want, you just need to keep in mind when it runs if you do that and that you have to take care of *everything* with regards to preconditions that are needed for your defined script to actually complete successfully.

drewzh, 2019-07-02:
Yeah, this definitely is due to me not understanding how these gcodes are processed. I just assumed that they'd operate in the same way as it you added them to the slicer gcode and that the gcode would be processed after the heat up sequence, but after your explanation it's clear that's not the case.

As the temperatures are not determined at the point of the Octoprint gcode starting (I'm assuming there's no variable system that could fetch the temperatures at this point), it would seem that trying to get Octoprint to heat up to the correct temperature and purge before the print starts wouldn't be doable at this stage.

I'll just accept this limitation and continue to send all my gcode from the slicer. I think it's worth adding a note to the top of the gcode section that explains some of the limitations, as for new users (and newcomers to 3d printing in general, it's a little confusing).

Thanks for your patience in explaining all of the above.

foosel, 2019-07-02:
I thought "Before print job starts" was already pretty clear to begin with, but apparently not, so 1.3.12 will have this:

[image]

Hope this helps.

drewzh, 2019-07-03:
Yeah that clears it up a bit, thanks. I think newcomers like me just don't fully grasp the sequence and at what stage the heaters fire up and that you can't use this section for any purge sequences etc without commanding the heaters manually, which isn't really possible without pre defining the temperatures in the octoprint config.

At least I'm clear on this now. Thanks for your time.

commit 44af0958 (2019-07-02): 🚸 Make clear when GCODE scripts are executed

Closes #3198
closed 2019-10-22 by foosel