OctoPrint/OctoPrint #3199: Unable to print any files after an mqtt checksum error

luketaverne, 2019-06-19:
#### What were you doing?

We click print on a file in the octoprint interface, in order to start a print job.

We submit a file to Octoprint via mqtt. There was a checksum error/missing checksum on one line (this is not the issue, just that an error occurred and we were unable to continue printing any files after this error, even after restarting). The printer gave a resend request which was not answered, and I assume that Octoprint created the `./octoprint/data/print_recovery_data.yaml` file to attempt recovery.

From the `octoprint.log` file:

```
2019-06-17 07:32:52,801 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 1550, current line = 1551
| Last lines in terminal:
| Recv: echo:busy: processing
| Recv:  T:199.95 /200.00 B:90.15 /90.00 T0:199.95 /200.00 T1:100.40 /100.00 @:21 B@:0 @0:21 @1:4
| Recv: echo:busy: processing
| Recv:  T:200.67 /200.00 B:89.56 /90.00 T0:200.67 /200.00 T1:100.94 /100.00 @:0 B@:0 @0:0 @1:0
| Recv: echo:busy: processing
| Recv: ok
| Send: N1545 G0 E0.052399226161743 F240*3
| Recv: ok
| Send: N1546 G2 X179.839143147134962 Y121.067724452220332 R0.1*64
| Recv: ok
| Send: N1547 G1 X179.921143147134956 Y120.986724452220329 Z0.81 E0.115260574352202 F240*104
| Recv: ok
| Send: N1548 G0 E0.051673566999117 F240*7
| Recv: ok
| Send: N1549 G2 X179.911497095951404 Y120.978145024319673 R0.1*73
| Recv: ok
| Send: N1550 G1 X182.802497095951423 Y117.271145024319665 Z0.81 E4.701034992424558 F240*103
| Recv:  T:200.49 /200.00 B:89.41 /90.00 T0:200.49 /200.00 T1:100.60 /100.00 @:3 B@:127 @0:3 @1:1
| Recv: Error:checksum mismatch, Last Line: 1549
| Recv: Resend: 1550
2019-06-17 07:32:52,812 - octoprint.util.comm - ERROR - Error while processing hook mqtt-controls for phase sent and command G1 X182.802497095951423 Y117.271145024319665 Z0.81 E4.701034992424558 F240:
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/util/comm.py", line 3064, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_mqtt_controls/__init__.py", line 149, in gcode_sent_hook
    'tags': sorted(tags),
TypeError: 'NoneType' object is not iterable
```
From here on, we were unable to print anything using Octoprint. It still responded to terminal commands, but clicking print on any file generated the following response in `octoprint.log`:

```
2019-06-18 06:21:04,978 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/api/files.py", line 369, in uploadGcodeFile
    filename = fileProcessingFinished(added_file, fileManager.path_on_disk(FileDestinations.LOCAL, added_file), target)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/api/files.py", line 340, in fileProcessingFinished
    selectAndOrPrint(filename, absFilename, destination)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/api/files.py", line 353, in selectAndOrPrint
    printer.select_file(absFilename, destination == FileDestinations.SDCARD, printAfterSelect, user)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/printer/standard.py", line 464, in select_file
    actual_origin = recovery_data.get("origin", None)
AttributeError: 'str' object has no attribute 'get'
```

We tried deleting the cache (`./octoprint/data/preemptive_cache.yaml`) and restarted the service (didn't work). We also noticed a file `./octoprint/data/print_recovery_data.yaml`, which seems to be related to the `recovery_data.get(...)` call in the last line of the previous error. We moved the file, restarted the octoprint service, and were able to continue printing as normal.

Contents of `print_recovery_data.yaml`:

```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for
```

#### What did you expect to happen?
Expected the file to print. Or, if there is an error in recovering a failed print, to see a message in the Octoprint interface or terminal. Additionally, if we reboot the system containing octoprint, that we are able to start a new print. This happened with any of the files we tried, not just the one that generated the error.

#### What happened instead?
The octoprint interface did nothing, the printer does not move, and there are no failure messages in the octoprint terminal. Unable to print any files.

#### Did the same happen when running OctoPrint in safe mode?

Unable to test in safe mode at the moment.

#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
Distributor ID: Debian
Description:    Debian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch

#### Printer model & used firmware incl. version
n/a

#### Browser and version of browser, operating system running browser
n/a

#### Link to octoprint.log
Posted above

#### Link to contents of terminal tab or serial.log
Posted above.

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s)/video(s) showing the problem:
n/a

I have read the FAQ.

foosel, 2019-07-29:
This looks like corrupted data, the contents of the file should look like this:

``` yaml
date: 1564411834.016
origin: local
path: foo.gcode
pos: 296
```

I have no idea what made plain GCODE show up in the file (maybe a plugin, maybe a dying filesystem), but in any case I've now added better content validation and hence hardening against issues like this caused by data corruption. Furthermore, I've added an additional check when the data is utilized on print start, so that should no longer cause issues there either.

Note that the printing issue that led to this, the quoted "mqtt checksum error" happened in a third party plugin that I have no control over.

foosel, 2019-07-29:
PS: Next time please attach a full `octoprint.log`, not just an excerpt. It's very important to get information about the full runtime environment that an error reproduces in, and that gets logged to that log file.

luketaverne, 2019-07-29:
Hi Gina,

Thank you for the feedback and the fix :). We know the third party plugin isn't your fault! And we're happy to hear there are now more measures in place to help recover when those plugins misbehave :)

Have a nice day, and thanks again!

Best,
Luke

commit 89255e2e (2019-07-29): ☔ Harden against corrupt recovery data

Closes #3199
closed 2019-07-29 by luketaverne