OctoPrint/OctoPrint #3201: [Request] MAC filtering (whitelist) feature

Radu13, 2019-06-20:
Hello i've been using this program for a while an i noticed i can literally go to the login page from every device just with IP and port.
For upgrading even more the security it would be good a mac whitelist settings where you first add the device's mac address and then you can access from them for remoting control the printer.

tedder, 2019-06-20:
Are you exposing your OctoPrint server to the internet? It should be on your local network only, and [anonymous users should not be allowed to access it](http://docs.octoprint.org/en/master/features/accesscontrol.html).

Radu13, 2019-06-21:
Not the octoprint one, but i'm hosting a public server with the same modem, if someone try random ports  could reach the login page, and maybe someone could enter.
But anyway it could be an useful and secure add-on.

tedder, 2019-06-21:
If it's on the same server, you should change what hosts Octoprint/HAProxy is listening for (say, 10.0.0.0/8 instead of 0.0.0.0/0) and firewall your ports. If you are DMZing your Pi, this isn't an Octoprint problem.

foosel, 2019-06-24:
Securing your network isn't OctoPrint's responsibility, firewalling software is better suited for this. Even if I put some MAC or address whitelisting into OctoPrint's webserver, no ones keeping any potential attacked from DOSing the target host. Use proper network security, don't port forward blindly.

closed 2019-06-24 by foosel

jubaleth, 2019-06-24:
To be really honest, I think you have a misconception about the security provided by MAC address filtering. I.e. there is *none*. Network traffic can easily be sniffed (while on the same network, and really, if you have OctoPrint connected to the internet, please stop....) and the MAC address can *very* easily be spoofed... Whitelisting is security by obscurity at best, and it's not even amazingly obscure... As @foosel pointed out, use a firewall to stop unwanted connections to your instance.