OctoPrint/OctoPrint #3203: [Request] API Documenation - Progress information

mdg357, 2019-06-24:
**Is your feature request related to a problem? Please describe.**
The API documentation for the Progress information lacks a description of the `printTimeLeftOrigin` field. It is unclear what values may be returned in this field.

**Describe the solution you'd like**
Document all possible values that may be returned by the `printTimeLeftOrigin` field.

**Describe alternatives you've considered**
N/A

**Additional context**
Specific documentation section found here: `https://docs.octoprint.org/en/master/api/datamodel.html#progress-information`

mdg357, 2019-06-24:
Seems like possible values _might_ be:
- linear
- estimate
- average
- analysis

foosel, 2019-07-29:
Fixed by the above commit, pushed to `maintenance`, soon `devel`, to be part of 1.3.12.

commit 372b3662 (2019-07-29): 📝 Add docs for printTimeLeftOrigin field

Closes #3203
closed 2019-10-22 by foosel