OctoPrint/OctoPrint #3204: [REQUEST] Add a close button to HOST_PROMPT_SUPPORT modal dialogs

drewzh, 2019-06-24:
**Is your feature request related to a problem? Please describe.**
When enabling HOST_PROMPT_SUPPORT in Marlin 2.0.x, Octoprint is able to capture and handle messages sent from the printer for events such as stopping the print from the printer side. Unfortunately the dialog that's displayed to the user is unclosable from mobile and it's also persistent between refreshes (which is fine if it were closable). This obviously causes functional issues. On the desktop browsers, you can press ESC, but that's not available as an option everywhere.

**Describe the solution you'd like**
Add the 'click outside dialog to close' behaviour and add a further 'dismiss/acknowledge/ok' button to the body of the dialog.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

foosel, 2019-06-24:
So... the thing is, this whole dialog business was originally intended to be modal because it should only be ever used by firmware in case of blocking operations that *require* user input and can't continue without it. Apparently some firmware features have regressed from that though, and we already had a ticket about that in #3171 which added the closing-by-ESC thing.

I guess since I've already had to give up on *enforcing the original intent behind this protocol addition that **I** designed* thanks to broken firmware out there and added the ESC functionality, I might as well add a closing button as well. Let's go all the way in abusing this functionality... 🤷‍♀️

foosel, 2019-06-24:
So I just checked and the dialog already has a close "X":

[image]

That also works on mobile (I just tested). I'll leave it at that.

Adding further "dismiss" buttons really goes *completely* against the whole idea of the interactive prompt system, see the discussion in the linked ticket above. "Click outside dialog to close" will be be possible to disable altogether in 1.3.12 btw since it was causing people too much trouble by accidentally dismissing dialogs they still wanted to use, see #3162.

closed 2019-06-24 by foosel

drewzh, 2019-06-24:
@foosel Can you please take a look at the below. This is using the very latest stable OctoPrint. I can not see a close button as you've shown above.

I understand however that this is meant to be a modal system - but for now, this feature seems incompatible with Marlins implementation.

I'll disable and forget about it for the foreseeable future until someone feels passionate about it enough to fix it in Marlin.

On Mobile:
https://lh3.googleusercontent.com/gMv37iBKeJcn-SADCl58vdhn8G31V_GReHCcsCny6UB9WpcN7Gpmx0K6D01J6Pb-DDxercZ3TY5Y4dMcIVE0oUmfjpQTkhBqsxyLl6ZyZ6A7WQVBsr-4DEzheHTZGwraTTAA_bOsbXQu6S1isd1YmSA29JIODXTM8QSmSHW4Y55ui0ZCnxDX-7Ay00A9wG6V6ORI99BIMiCdYE8Gb3GoHtelZ3XmDLHYJ2yIMRQgfn88mybbNrDIUaf0a2QJdaEsIcjqgP324EPxcR7nKO-BvjgleEo-XKuJwPX1fQgShc24d244PUoF2G5oWRESbv9aaXbxN1Q1NiXvZ22ppweEBQxTDEul_LqqBMvsSE4modOKExAkQ-nDo0LvyIyLi5Gwkk6WPop6mviR8nKXmcWHry5Gui7VOjxX6UE3d44s7DlvTqTpJ2nNMBVQrVbS7PFRh3Z632Nu6bvCL-5eqGAzjv2UrvzkW_f0GSXIDpgQ6JZANY8p4qQYLFDNLo8DFsNsYTkFJsKVkDMWcLws7POuS-f-OeoX_TdCBw9W0sFG_FMXwpdT6FUFkZljKOVCCEoUIu1mVLm9ncTjbxkXbB0vvC_2IsZbqGQVV2UMxpwfzRMVVaWmiDP4NCbc-kBeLxd9DDzgZ764Lo6VjZZScIb-7ofQ9eIlQVTZ=w848-h1694-no

On Desktop (Latest Chrome stable)
https://lh3.googleusercontent.com/wVb48-pqGpNIDRg1_rAwUiOH79Oqxepcybu2TKjAqg7pNa2fO2UCfCuYa8jKc-n4TquXS2wZq_uwMqEOgCsbzMM3nSGJHC_wmPbd5b0cldWA6JWEG7MqWzkDUSifL_p1pDPo3tj0fJSW9wewOLSahfdOa90kWJ8TLa9ZwB_V2NNo1VdyKF0H8iNGSRfPfaKD9aRUf0MR5HLtjjQabJCl8OkdW6pGNfKadTPo6XgcAAMTYwb0nD25Pll9t-2lYGGOiMhdSr-PAtgbml5Usju-8SlDFf2cRS1OzH_O5OI76Dh6SjZ2cRIyaR_ud6h5Rg7hEhQrFaxS17dfohQrInCj810R7QirrKgEA6VVdiX4IJ5ykHJi7m_ueYH6W-YNJ_QQLNB_z66rsSVTJvl58uwEXBY7b3p-GT-FXM0OwosI_hH6RsBQ40camCfPUJ1CbetsBqdtSTMuHi4FRCywh7yX1qsdJ1uImKHIqGDMQ7iYARIi5WdAg0hBJVW_0hwhwksEi62YDMzk-Yi-GCk8c4U8gnb7CGwzPljvvFYv08CMvw1987qqC7tWR5TlZ3wPgckU3TI8S8trakTXYzmwFA0HYo3KK4RU_qk17jVNrMftFI3mEEH2YZfc0tkXxHUC5ziQMtrb5MkWHo_f-PW6OkfQvZBxhatkk12L=w1988-h1208-no

Element inspection showing no hidden close button:
https://lh3.googleusercontent.com/xzDr1AhgO4rip1tO7Ovm03LlFMPlAF_eUv1mWiL3u6cP2NYB4HEKUnzC5ADxNcjtpQyDdWAYRd9ehvSyDjUR2UIWuCmkt-BE1zRJqk9h1bMHOnNgyRjz9eZo72C54PpmeQHE682D3wShlUs2cuvsiulgWZQuNhEqgSXI74qSRx7ZTKZF-cVXeI7w6OHJ12BKiaFQsRuvyp2aoTvUULXeGFFZwSWWVFymTzFM6a3A_p-TYnABv36AiRNI6ZRaib7vugZnpmLHK8UMdERBXaRvErD-WbzpW5aDkj6OBJH52ubcKER6uhhXED46gtqCNYO_KZsB3HfH68WfzKaV3koQdMzdclO8SzmdbK5onn7_Qrkf_RVMWV_ZNpVtpZIpNGZsi7NIRbHrm-AaQ2uc-cx-KL0sFfCkdbjDi24D0Vuyzswf0V6ttFg74aBjDIqAjRcs1aD9qgaddgHNgXd8ewGrme6VNaVu0M92t7o3Er8pcW0EWIfeB64oXM8F65fXC8eic0RvmmEkvCBpdkXKWx_nlP_IGd-gg1BWVmAizEWTvMSjrt6Q-2woAmHCP6kyQVeojV6_WHiIHH0QasCL7Tl9ehKTODTQ1YJNYKahZ4aZqbAE_SQbFoQQ_CUJuNt50qHZET6mLW8KLuf_TpLzF_6NKfX28P1Vp-jE=w2140-h1146-no

foosel, 2019-06-24:
The links to the screenshots sadly don't work.

drewzh, 2019-06-24:
Ah apologies. Try these...

https://photos.app.goo.gl/1QsmmF2a5PHvuChq5
https://photos.app.goo.gl/48GZ2R9asNwKzrEF8
https://photos.app.goo.gl/WgETNh7Bkp4bTQgL7

foosel, 2019-06-25:
Nope 😂

[image]

drewzh, 2019-09-11:
@foosel Hi. Were you able to reproduce the issue with the modal? The close button is not visible in the browser (though clickable).

foosel, 2019-09-11:
@drewzh I could see none of the pictures you posted, so I didn't even know what was up and hence haven't been able to look into it.

drewzh, 2019-09-11:
@foosel Please let me know if this works. I just created an account and uploaded the above images to imgur.com:
https://imgur.com/a/DbXd1Q3

foosel, 2019-09-11:
What does it say in the lower left corner of the web interface?

drewzh, 2019-09-11:
@foosel Unfortunately I'm not able to reproduce the issue where I currently am as I'm at work, away from my printer. I'll re-enable prompt support and re-asses.

Do you mean the bottom left of the Octoprint interface. This was captured with the latest Octopi stable and OctoPrint 1.3.11, but I'm currently using the maintenance release, to get the changes for M112 fix.

foosel, 2019-09-11:
> This was captured with the latest Octopi stable and OctoPrint 1.3.11

OctoPrint 1.3.11 didn't yet contain the changes. 1.3.12rc1 and `maintenance` do.

drewzh, 2019-09-11:
Ahhhhh, that explains that then :D I'll re-enable later on and start using the feature again. Thanks Gina!