OctoPrint/OctoPrint #3206: Octoprint not connecting in startup via GPIO

gusdleon, 2019-06-24:
#### What were you doing?

I have a Original prusa MK3 connected a Raspberry Pi via GPIO, and the Raspberry running the latest octopi, and configured to auto connect on startup, but the printer always responds with garbage OK the in the first connection

#### What did you expect to happen?
When I turn on the printer to the Octoprint server starts automatically connect to the printer via the GPIO serial connection and then display the IP in the printer display

#### What happened instead?
When I turn on the printer, the Octoprint server starts tries to connect to the printer, but its fail due timeout because the garbage "ok" of the printer
#### Did the same happen when running OctoPrint in safe mode?

I didn't tried at this time, only occurs in printer startup.

#### Version of OctoPrint
1.3.10 replicable in 1.3.11 too

#### Operating System running OctoPrint
OctoPi 0.16.0

#### Printer model & used firmware incl. version
Original Prusa i3 MK3S MULTI MATERIAL 2S
Firmware 3.7.1

#### Browser and version of browser, operating system running browser
N/A
but latest chrome in latest windows 10

#### Link to octoprint.log
Clear octoprint.log created on printer and raspberry with octoprint startup 
https://gist.github.com/gusdleon/e1f41d79aa14f90282ce7f940eeac72f
Clear octoprint.log created after clearing the past log and rebooting system via web interface 
https://gist.github.com/gusdleon/1b8a23c715d1fe6d91921c38786e84de

#### Link to contents of terminal tab or serial.log
Clear serial.log created on printer and raspberry with octoprint startup https://gist.github.com/gusdleon/9c60f3c383bb02fb08258c8499a1e02b
Clear serial.log created after clearing the past log and rebooting system via web interface 
https://gist.github.com/gusdleon/56fbcc270ac181790cd855d3babc99e3

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
Sorry no videos/screenshots :(

I have read the FAQ. ✔️

foosel, 2019-07-02:
> but the printer always responds with garbage OK the in the first connection

I'm a bit confused that this garbage doesn't show up in the log you shared. Is it visible in the terminal tab? If so could you share a screenshot of that?

foosel, 2019-07-02:
Ah, never mind, I found it in the raw log file. The web interface just doesn't display it.

Not sure I can do anything about this. If the printer sends garbage data on first connect, that's the way it is. The only thing I could try is send a second `M110` here "just to make sure" and see if that produces a clean `ok` in response...

foosel, 2019-07-02:
Can you check please if the problem goes away if you check "Wait for `start` on connect" under Settings > Serial > Firmware & protocol > Protocol fine tuning?

gusdleon, 2019-07-03:
> Can you check please if the problem goes away if you check "Wait for `start` on connect" under Settings > Serial > Firmware & protocol > Protocol fine tuning?

Hey Gina Thank You for your help ❤. 
Trying with this the result is another timeup.. but in this case the pi miss the start message for the longer boot time.

Let Me explain...
I have a Original Prusa i3 MK3 with the Einsy board, and this board as you already know have a port to connect a pi zero directly via gpio, so the pi turns on at the same time of the printer, ( yes I will upgrade the zero for my 3B+ later), and because the printer boots up before the pi, the pi misses the "start" message from the printer

I really want to connect the printer via this method (GPIO) because the printer dont restarts in connection with octoprint, and with this method I can start a print via the printer display, and later login via the web interface without disrupting the ongoing print.

Im trying [This method](https://github.com/foosel/OctoPrint/issues/1945#issuecomment-317333247) via usb, but didn't work, in the printer/pi boot the printer restart to make the first connection

Aaaaand, how do I can update my Octoprint installation to the branch that includes the commit you maked referring to this issue, 

I would like to help testing :)

Thank You again :D

Regards.

gusdleon, 2019-07-03:
> Aaaaand, how do I can update my Octoprint installation to the branch that includes the commit you maked referring to this issue,
> 
> I would like to help testing :)

never mind that.. I found the [Guide](https://community.octoprint.org/t/how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image/206) to update to the maintanance branch and I started testing I keep you posted about this 😉

[image]

Thank you again Gina 😁

gusdleon, 2019-07-09:
Hey Gina, me again, after 6 days of testing, all the problems are gone...

The octoprint connects to the printer on startup, I can send jobs directly from the slicer without intervention, even I can restart the entire pi and connect it to the printer again without affecting current jobs, (when printing directly from sd because the connection via GPIO).

The problem with the first ok stills as you can see in the next screenshot
[image]

But its problem of the firmware for sure, I have already created a Issue in the repo of Prusa Firmware, but without any answer since I opened 10 days ago... But it's no problem because after the connection is established no more problems are detected..

I attached the new serial.log and octoprint.log if you need it. 
#### Link to octoprint.log ([f2da9d8](https://github.com/foosel/OctoPrint/commit/f2da9d8d7d66942271668792028e716eb053e9b3))
https://gist.github.com/gusdleon/53c2b432d5644658310ca81c269bac65

#### Link to serial.log ([f2da9d8](https://github.com/foosel/OctoPrint/commit/f2da9d8d7d66942271668792028e716eb053e9b3))
https://gist.github.com/gusdleon/950ac73a9fc7fe65392422e363fbf196

Thank you so much Gina <3

commit 85d212ba (2019-07-02): ☔ More resilience against garbage on the line on initial connect

We'll now send the hello command twice, and a third time if we run into
a timeout during initial connect.

Closes #3206
closed 2019-10-22 by foosel