OctoPrint/OctoPrint #3207: Timelapse Custom Snapshot URL doesn't work if Server gives no Content-type header

LinoBarreca, 2019-06-25:
1. Go to Settings - Webcam & Timelapse - Timelapse Recordings - Snapshot URL
2. Insert a custom Snapshot url (I tried with UV4L snapshot url http://127.0.0.1:5001/stream/snapshot.jpeg)
3. Click on Test
4. Error appears: "Could retrieve the snapshot URL, but it didn't look like an image. Got this as a content type header: text/html Please double check that the URL is returning static images, not multipart data or videos."

Actually, seeing the RAW request/response UV4L server does NOT include a content type header at all but just
HTTP/1.1 200 OK
Date: Tue, 25 Jun 2019 08:11:10 GMT
Content-Length: 388905

#### Version of OctoPrint
1.3.11

foosel, 2019-07-01:
>     4\. Error appears: "Could retrieve the snapshot URL, but it didn't look like an image. Got this as a content type header: text/html Please double check that the URL is returning static images, not multipart data or videos."

If you really got back that exact message, then your quoted response from the server can't possibly be the correct one.

OctoPrint would have thrown a different error if there had been no `Content-Type` header in the first place. It doesn't just blindly assume `text/html` - that looks rather like you actually gave it the URL of a website.

In any case, the above commit adds mime type guessing in case no `Content-Type` header is supplied on snapshot test. Will be part of 1.3.12.

(PS: Next time *please* fill out the ticket template, it isn't there for fun 😕)

LinoBarreca, 2019-07-01:
You are right. And I am sorry. I didn't fill them all because, as a programmer, I thought I gave everything necessary to solve the bug without wasting unnecessary time. Next time I will include also all the other informations.

"OctoPrint would have thrown a different error if there had been no Content-Type header in the first place. It doesn't just blindly assumetext/html - that looks rather like you actually gave it the URL of a website."
Then you really should double check something because he did assume text/html when he should have thrown a different error at all. I am 100% sure the url was correct because I directly dumped the data from the server itself. It contained the above header and the data only.

foosel, 2019-07-02:
> Then you really should double check something because he did assume text/html when he should have thrown a different error at all.

I wouldn't have written the above if I hadn't already double checked that ;) I wrote a small server that replicated the response you provided and checked the behaviour with that and observed something different from what you shared.

LinoBarreca, 2019-07-09:
I tried again, the url was correct and the message was what I said. If you want I can give you a test url, so you can see it yourself both the response and the error from octoprint....
[image]

Also the "live cam" doesn't work if the stream isn't mjpeg but h264. I guess because the answer never ends.
To be honest Firefox has a weird behavior too when opening the h264 url (DOS), I will signal it to Mozilla. Chrome instead works just fine.
(I can provide the url for that as well)

foosel, 2019-07-09:
> I tried again, the url was correct and the message was what I said. If you want I can give you a test url, so you can see it yourself both the response and the error from octoprint....

That would be helpful.

> Also the "live cam" doesn't work if the stream isn't mjpeg but h264.

That's to be expected, OctoPrint currently only supports mjpg streams, not h264.

LinoBarreca, 2019-07-10:
> > I tried again, the url was correct and the message was what I said. If you want I can give you a test url, so you can see it yourself both the response and the error from octoprint....
> 
> That would be helpful.

http://94.34.2.35:57731/uv4l/stream/snapshot.jpeg

ip is dynamic so it's valid now.

> > Also the "live cam" doesn't work if the stream isn't mjpeg but h264.
> 
> That's to be expected, OctoPrint currently only supports mjpg streams, not h264.

Why should octoprint even care about the format of the video stream?
if the user configures a custom url just show the url in an iframe, without caring about what's inside.

foosel, 2019-07-11:
> http://94.34.2.35:57731/uv4l/stream/snapshot.jpeg
> 
> ip is dynamic so it's valid now.

I just tried with this URL against a fresh OctoPrint 1.3.11 and got the error response I expected based on the implementation, not what you reported:

[image]

Headers are identical to what you reported however:

``` 
$ curl -I http://94.34.2.35:57731/uv4l/stream/snapshot.jpeg
HTTP/1.1 200 OK
Date: Thu, 11 Jul 2019 09:12:22 GMT
Content-Length: 319061
```

> Why should octoprint even care about the format of the video stream?

Because the current implementation relies on an `img` tag and not a `video` tag.

> if the user configures a custom url just show the url in an iframe, without caring about what's inside.

That's an existing feature request, and certainly not a case of "just" due to various factors like existing third party clients, plugins and such that rely on the current implementation and would break if that was to be changed in a backwards incompatible matter.

LinoBarreca, 2019-07-11:
> [image]
> 
> Headers are identical to what you reported however: 
> ```
> $ curl -I http://94.34.2.35:57731/uv4l/stream/snapshot.jpeg
> HTTP/1.1 200 OK
> Date: Thu, 11 Jul 2019 09:12:22 GMT
> Content-Length: 319061
> ```

This is so weird. On my end it assumes text/html while on yours it says it can't even fetch it.
Could it be because my Octoprint runs on Octopi 0.16? (just wondering)
Where are you doing these tests?
Could that be something related to the python version installed?
I don't know....however even "Could not retrieve snapshot url" is still a wrong message if you actually can but the answer is not what you expect (the message I get makes more sense. Basically it's: I retrieved it, but it's not an image...which is partially true since the server doesn't explicitly say it is)

> Because the current implementation relies on an `img` tag and not a `video` tag.
Fair enough.

> That's an existing feature request, and certainly not a case of "just" due to various factors like existing third party clients, plugins and such that rely on the current implementation and would break if that was to be changed in a backwards incompatible matter.

Fair enough too. However mjpeg support should be dropped at all since it consumes a lot of bandwidth and CPU resources. h264 is done in hardware and consumes no resources at all (10% cpu on just core 1 on a pi 3b while streaming and printing, checked through ssh and htop)

foosel, 2019-07-11:
> However mjpeg support should be dropped at all since it consumes a lot of bandwidth and CPU resources

On the client, yes. On the server, no. The mjpg choice was made back when it was made because a) hardware accelerated server side h264 encoding was still pretty much impossible to get right on a Pi (and it's still a bit of a mess today) and b) mjpg is usually supported natively by even the cheapest USB webcams and hence doesn't cause any kind of CPU overhead on the encoding server.

> Could it be because my Octoprint runs on Octopi 0.16? (just wondering)

That was also on OctoPi 0.16, so no.

> Where are you doing these tests?

Pi 2 on my LAN running OctoPi 0.16 with OctoPrint 1.3.11. 

> Could that be something related to the python version installed?

No. If at all it could be related to `requests`, but that should be the same version. 

> however even "Could not retrieve snapshot url" is still a wrong message if you actually can but the answer is not what you expect

Which is why the above commit added content type guessing, so this particular situation won't arise anymore starting with 1.3.12 anyhow.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel