OctoPrint/OctoPrint #3210: [BUG] event "Error" not firing

aschor, 2019-06-28:
#### What were you doing?
Nota : in settings -> serial connection -> error handling,   I have 2nd choice enabled (cancel print etc. etc.)

Trying to get an event subscription in config.yaml to shutdown relays in case the printer reports an error. Here is what I have : 
events:
  enabled: true
  subscriptions:
  - command: python ~/scripts/relaysinit.py && python ~/scripts/relaysOn.py
    enabled: true
    event: Startup
    type: system
  - command: python ~/scripts/relaysOff.py
    enabled: true
    event: Shutdown
    type: system
  - command: python ~/scripts/relaysOff.py
    enabled: true
    event: Error
    type: system

1) When I start the system, the relays switch On
2) when I stop the system, the relays switch Off
3) But when I try to force an error, for example I launch a print without pluging in the heatbed :

the printer reports an error
But not "Error" event is triggered whatever I try. Even the logs don't mention the event :

2019-06-28 15:59:30,502 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2019-06-28 15:59:30,513 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: fan_pcb2.gcode, owner: arnaud, user: arnaud
2019-06-28 15:59:30,535 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-06-28 15:59:30,547 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2019-06-28 16:00:32,703 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Heater heater_bed not heating at expected rate
| Last lines in terminal:
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: B:27.3 /95.0 T0:26.8 /0.0
| Recv: // Klipper state: Shutdown
| Recv: ok
| Send: N8 G21*18
| Recv: ok
| Send: N9 G90*25
| Recv: // Heater heater_bed not heating at expected rate
| Recv: // See the 'verify_heater' section in config/example-extras.cfg
| Recv: // for the parameters that control this check.
| Recv: //
| Recv: // Once the underlying issue is corrected, use the
| Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
| Recv: // config, and restart the host software.
| Recv: // Printer is shutdown
| Recv: !! Heater heater_bed not heating at expected rate
2019-06-28 16:00:32,710 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2019-06-28 16:00:32,739 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: fan_pcb2.gcode, owner: arnaud, user: None
2019-06-28 16:00:32,792 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Heater heater_bed not heating at expected rate
| Last lines in terminal:
| Recv: //
| Recv: // Once the underlying issue is corrected, use the
| Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
| Recv: // config, and restart the host software.
| Recv: // Printer is shutdown
| Recv: !! Heater heater_bed not heating at expected rate
| Changing monitoring state from "Printing" to "Cancelling"
| Send: N10 M108*27
| Send: N11 M84*47
| Recv: ok
| Send: N12 M104 T0 S0*18
| Recv: // Heater heater_bed not heating at expected rate
| Recv: // See the 'verify_heater' section in config/example-extras.cfg
| Recv: // for the parameters that control this check.
| Recv: //
| Recv: // Once the underlying issue is corrected, use the
| Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
| Recv: // config, and restart the host software.
| Recv: // Printer is shutdown
| Recv: !! Heater heater_bed not heating at expected rate
2019-06-28 16:00:32,823 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: fan_pcb2.gcode, owner: arnaud, user: None

#### What did you expect to happen?
I did expect to see in the logs something like that : 
2019-06-28 14:50:46,763 - octoprint.events - INFO - Processing startup event, this is our first event
2019-06-28 14:50:46,763 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2019-06-28 14:50:46,771 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-06-28 14:50:46,772 - octoprint.events - INFO - Executing a system command

and I expected the "Error" event to trigger, my relays to switch off

#### What happened instead?
nothing

#### Did the same happen when running OctoPrint in safe mode?

YES

#### Version of OctoPrint

OctoPrint 1.3.11 running on OctoPi 0.16.0

#### Operating System running OctoPrint

OctoPrint 1.3.11 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version

Klipper latest

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

https://gist.github.com/aschor/67c190ad4a1812577c2b8dc0ba3a73d5

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-07-01:
It looks like you have this configured:

[image]

and the firmware doesn't report a fatal error here based on the described criteria.

In that case the error event isn't triggered either and this behaviour is correct (albeit not desired for your workflow).

aschor, 2019-07-01:
Hi @foosel , thanks for reviewing this.

I had supposed that "what to do on a firmware error (Error: or !!)" would fire an error event on ANY error received by the printer, that is to say when receiving what is specified : lines starting by "Error:" or "!!" (the left of the picture)

But what you're saying is that the error event is only triggered where receiving "fatal:" or "kill() called" on the right on the picture ? So "error" event would then be mislabelled, and be more a "fatal" event

Do you know any way to trigger "actions" on "!!" or "Error:" lines then please ? since actions plugins works on ... action

foosel, 2019-07-01:
To be honest, I'm not entirely sure if the `Error` event behaviour should stay this way... but it's a bit of a tricky decision. 

The `Error` event was introduced at a time when any kind of non-communication error was considered fatal and triggered a disconnect from the printer. When the new toggles where added to change that behaviour, the `Error` event was treated the same way in case anyone had an automation or something attached to it that should only trigger on fatal errors, so effectively to mirror the behaviour of OctoPrint. 

Of course there's also a perfectly valid reason to expect the `Error` event to always be triggered when an error line is received from the firmware, and to rather trigger something like a `FatalError` in case of actual fatal errors.

With backwards compatibility and an unknown number of existing installations possibly relying on the current event behaviour out there however, this is not easily solved without breaking stuff for people in the field 🤔

Part of me just wants to either fire the `Error` event in all cases or introduce a `FatalError` event. But as I said, that might break things.

foosel, 2019-10-31:
I've decided to leave things as they are. Any changes here might have unforseen side effects on solutions outside of my control (plugins, clients, ...).

I've adjusted the documentation slightly to reflect the limitations of the `Error` event.

If you need to react to (or want to handle) error messages from a plugin, you can take a look at [this hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-error).

commit 888fb97e (2019-11-08): 💡 Clarify Error event in docs

Closes #3210
closed 2019-10-31 by foosel