OctoPrint/OctoPrint #3211: Wizard window keeps on popping up on until hard-refresh

kennethjiang, 2019-07-01:
This issue is based on multiple reports I received from my plugin users. I have started to receive such reports since 1.3.11 was released.

#### What were you doing?

1. Install OctoPrint Anywhere plugin
2. Restart OctoPrint
3. OctoPrint Anywhere setup Wizard pops up
4. Click finish
5. Check `config.yaml` and confirm that `anywhere: 4` in `seenWizards` section
6. Restart OctoPrint again
7. Refresh OctoPrint page.
8. Bug: OctoPrint Anywhere setup Wizard pops up again.

#### What did you expect to happen?

OctoPrint Anywhere setup Wizard shouldn't pop up after the 1st time.

#### What happened instead?

OctoPrint Anywhere setup Wizard keeps on popping up on each OctoPrint restart.

Hard-refreshing OctoPrint page will stop the wizard window from popping up again.

#### Did the same happen when running OctoPrint in safe mode?

No. But OctoPrint Anywhere plugin is not loaded in safe mode.

#### Version of OctoPrint

1.3.11

#### Operating System running OctoPrint

Raspian

#### Printer model & used firmware incl. version

I don't have to connect to any printer to reproduce this bug.

#### Browser and version of browser, operating system running browser

Google Chrome Version 75.0.3770.100 (Official Build) (64-bit). Mac OS X.

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/3345846/octoprint.log)

#### Link to contents of terminal tab or serial.log

N/A 

#### Link to contents of Javascript console in the browser

[octopi2.local-1561989020300.log](https://github.com/foosel/OctoPrint/files/3345871/octopi2.local-1561989020300.log)

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

jneilliii, 2019-07-01:
I have had similar reports for my BeLevelVisualizer plugin (https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/133) that was resolved by restarting the browser and forcing a cache refresh.

foosel, 2019-07-01:
Hmm... sounds possibly related to #3166 which also is a wizard dialog that won't go away until a force refresh. Sadly I so far have not been able to reproduce that one myself.

foosel, 2019-07-02:
... of course now that I have an idea what it could be, I can reproduce reliably and am utterly confused that it ever worked in the first place.

foosel, 2019-07-02:
After I finally figured out what's going on I've pushed 72167d14d2796e8527f09d183107fad99c0ab2fa which should fix this. I also pushed 59ae7e39f229310dab11ebf15c8b2959decd3aab and 530fca95e49000bead67bf677844f5f80a037b1f for some additional safeguarding against issues like this.

Pushed to `maintenance`, soon `devel`, to be released with 1.3.12.

Until then, a forced refresh (holding shift & clicking on the reload button) should solve this.

kennethjiang, 2019-07-02:
Thank you so much @foosel  for your prompt action!

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel