OctoPrint/OctoPrint #3213: Octoprint crashes upon resuming print

diyengineer, 2019-07-02:
-->

#### What were you doing?
While printing, clicking pause, results in a normal pause. When "resume" is pressed the entire octoprint crashes (goes offline).

1. Start a print.
2. Click on pause after its begun
3. Once its time to resume click resume
4. See error as print server goes offline.

#### What did you expect to happen? 
Expected to pause a print, and go through my normal resume routine. Instead it resumed, and the octoprint server crashed. Repeated across 2 printers multiple times, all result in same problem.
#### What happened instead?
octoprint server crashed and doesn't really appear to record in either of the log file what happened.
#### Did the same happen when running OctoPrint in safe mode?
Will try first thing tomorrow. 

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
.16

#### Printer model & used firmware incl. version
Solidoodle press + Marlin flavor

#### Browser and version of browser, operating system running browser
Windows 10, Chrome
[octoprint (2).log](https://github.com/foosel/OctoPrint/files/3348324/octoprint.2.log)
[octoprint (1).log](https://github.com/foosel/OctoPrint/files/3348325/octoprint.1.log)
[serial.log1.txt](https://github.com/foosel/OctoPrint/files/3348328/serial.log1.txt)
[serial.log.2.txt](https://github.com/foosel/OctoPrint/files/3348329/serial.log.2.txt)

#### Link to octoprint.log
Attached

#### Link to contents of terminal tab or serial.log
Attached

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-07-02:
> #### Did the same happen when running OctoPrint in safe mode?
> 
> Will try first thing tomorrow.

Please report back when this is clarified. It doesn't make sense to look into an issue if it's not verified that it's reproducible with *stock* OctoPrint.

diyengineer, 2019-07-03:
Foosel,

We tested 3 machines in safe mode, and the problem vanished! We flashed the card and installed a new copy, ran perfect without safe mode! As soon as we installed: 

https://plugins.octoprint.org/plugins/filamentencore/

The problem came right back once the plugin was installed. It pauses as intended, but as soon as "resume" is clicked this appears and i have to power cycle the printer.
[image]

I will try your instructions and report back _tomorrow:_

foosel, 2019-07-03:
Hm, that sounds like there's an issue with this particular plugin then. I'm going to close this ticket, however it would still make sense to run the tests (as you have planned anyhow) and report your findings in the ticket on the plugin's tracker that you already opened (👍).

I'll also disable the plugin on the repository for now until this issue is clarified.

closed 2019-07-03 by foosel

coolcheat, 2019-07-28:
Just thought I'd chime in, I'm a frequent user of filament sensor encore and have never had a crash when resuming / pausing...