OctoPrint/OctoPrint #3216: Rendering of timelapse  failed with return code 255

Lino77, 2019-07-03:
[image]
what is return code 255

foosel, 2019-07-03:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2019-07-03 by foosel

pfanntec, 2019-07-19:
Hi @Lino77,

Check the webcam settings if the path to FFMPEG is ok.

For me (Raspbian Buster Image) _**/usr/bin/avconv**_ was the wrong path.
I changed it to **_/usr/bin/ffmpeg_** and now it works.

vtjballeng, 2020-02-06:
> Check the webcam settings if the path to FFMPEG is ok.
> 
> For me (Raspbian Buster Image) _**/usr/bin/avconv**_ was the wrong path.
> I changed it to **_/usr/bin/ffmpeg_** and now it works.

I couldn't update plugins with older versions of OctoPi anymore so I imaged 0.17.0 and restored backups from my older installations. Everything worked except the above issue and this comment saved my day. Thanks @pfanntec !