OctoPrint/OctoPrint #3220: Typo "_log_running_command" for comm.py property

Mrnt, 2019-07-06:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-07-08:
Thanks for the catch, should be fixed on `maintenance`, soon `devel` and to be part of 1.3.12.

Mrnt, 2019-07-18:
Thanks for all your hard work on this project!

commit d44747b3 (2019-07-08): 🐛 Fix long_running_command flag for M191

Closes #3220
closed 2019-07-18 by Mrnt