OctoPrint/OctoPrint #3224: Handling of open file and disconnects

i0n0s, 2019-07-08:
#### What were you doing?

1. Connect to your printer
2. Open a gcode file for printing
3. Disconnect your printer
4. Reconnect your printer
The file is still selected as active, but you're not able to click on the print button. In the 
file section, the "load and print" button can be selected but has no effect, the "load" button is greyed.
To be able to print the file again, you need to switch to another file and back again.

#### What did you expect to happen?
Click on the "Print" button to start the print.

#### What happened instead?
The button is inactive and can't get clicked.

#### Did the same happen when running OctoPrint in safe mode?
Yes, exactly the same behaviour.

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
Linux OctoPi 4.14.34

#### Printer model & used firmware incl. version
Printrbot Simple Metal
Marlin 1.1.9

#### Browser and version of browser, operating system running browser
Opera 60.0.3455.151, Windows

#### Link to octoprint.log
https://pastebin.com/V6sQMNax

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-07-16:
Hm, I can't reproduce this.

I tried the following:

  1. Connected to the printer
  2. Selected a file for printing
  3. Clicked the disconnect button, at which point the file selection cleared 

I also tried this:
  
  1. Connected to the printer
  2. Selected a file for printing
  3. Simulated a hard disconnect of the printer, at which point the file selection cleared

Can you reliably reproduce this? Can you provide more detailed reproduction steps?

i0n0s, 2019-07-16:
I can reproduce it but I'm missing a detail:
The file selection is cleared, at least in the status. But internal the file is still selected. I've tried to reproduce it again and at step 3, the file analysis was not finished and continued to run. But the filename was removed from the status. After reconnecting the "Open" button in the file section is greyed and in the status is a layer count visible.
Please add step 4: "Connected to the printer" and then check the status of the file in the file section, not in the status.

foosel, 2019-07-17:
Ok, I can reproduce this. Looks like a purely frontendside issue, if I reload the page or refresh the file list the file becomes unselected.

foosel, 2019-07-29:
Should be fixed by the above commit and will be part of the 1.3.12 release.

Already on `maintenance`, soon `devel`.

i0n0s, 2019-07-29:
Many thanks.

commit e007dbaf (2019-07-29): 🐛 Fix file selection not being cleared on disconnect

Closes #3224
closed 2019-07-29 by i0n0s