OctoPrint/OctoPrint #3226: MK3s NOT working... tried everything...

zarigata, 2019-07-13:
i have an prusa mk3s and a raspberry pi 3b connected with him, but ive tried to connect everytime its giving me this error

Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x6a51e5f0, open=True>(port='/dev/ttyACM0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 9600
Send: N0 M110 N0*125
Recv: Command not found!
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 115200
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 250000
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 230400
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 57600
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 38400
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 19200
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Changing monitoring state from "Detecting baudrate" to "Error: No more baudrates to test, and no suitable baudrate found."
Changing monitoring state from "Error: No more baudrates to test, and no suitable baudrate found." to "Offline (Error: No more baudrates to test, and no suitable baudrate found.)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x6a461610, open=True>(port='/dev/ttyACM0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 9600
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 115200
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 250000
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 230400
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 57600
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 38400
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Trying baudrate: 19200
Send: N0 M110 N0*125
Baudrate test retry #1
Send: N0 M110 N0*125
Baudrate test retry #2
Send: N0 M110 N0*125
Baudrate test retry #3
Send: N0 M110 N0*125
Baudrate test retry #4
Send: N0 M110 N0*125
Changing monitoring state from "Detecting baudrate" to "Error: No more baudrates to test, and no suitable baudrate found."
Changing monitoring state from "Error: No more baudrates to test, and no suitable baudrate found." to "Offline (Error: No more baudrates to test, and no suitable baudrate found.)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x6a514490, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Recv: start
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x6a4c62f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Recv: start
Send: N0 M110 N0*125
Connection closed, closing down monitor
Changing monitoring state from "Connecting" to "Offline"
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6b856050, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Recv: start
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6b856450, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Recv: start
Send: N0 M110 N0*125
Connection closed, closing down monitor
Changing monitoring state from "Connecting" to "Offline"

lf-, 2019-07-14:
Set your baudrate to 115200. If still not working, try another USB cable.

marcboivin, 2019-07-14:
The MK3S can be hooked up to the pi headers too.

closed 2019-07-16 by foosel