OctoPrint/OctoPrint #3228: Full RX Buffer - Occational 1-3 second pause

FormerLurker, 2019-07-18:
#### What were you doing?
Printing a gcode file.  I noticed a pause of about 1-3 seconds, and immediately checked the octoprint.log file.  Shortly after this, I experienced another pause.  Both logs will be supplied later.

I also filed an issue on Prusa's firmware repo [here](https://github.com/prusa3d/Prusa-Firmware/issues/738#issuecomment-512896087).

Very difficult to reproduce.  I've only seen it happen a few times.  In the past I blamed my plugin for the issue, but this time I confirmed that it happened in the middle of the layer, nowhere near where my plugin operates.

#### What did you expect to happen?
That the 'Full RX Buffer' response could be handled faster.

#### What happened instead?
There was a 1-3 second delay (closer to 2)

#### Did the same happen when running OctoPrint in safe mode?
I will update the issue if I notice this.  I have to run several prints for comparison without any plugins enabled, and will report back once this is complete.  I don't expect you to respond until that takes place, but it's easier for me to create the issue this way.

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
Rasbian

#### Printer model & used firmware incl. version
MK2.5 w mmu - firmware 3.7.1

#### Browser and version of browser, operating system running browser
NA/ not a browser issue

#### Link to octoprint.log
https://gist.github.com/FormerLurker/e051747cf467df82491917a93e795912
(Please ignore the unhandled exceptions from Octolapse, I believe they are unrelated)

#### Link to contents of terminal tab or serial.log
The related terminal log is contained within the octoprint.log, but I will attempt to get a serial log in the coming days.

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s)/video(s) showing the problem:
NA

I have read the FAQ.

reloxx13, 2019-08-12:
like this?
https://github.com/foosel/OctoPrint/issues/2799

FormerLurker, 2019-08-12:
No, I don't think it's the same issue.  I'm starting to suspect linear advance.  I usually turn this off when using Octoprint, but noticed it was on after updating my slicer profile.  I'm still doing some testing to verify.

Thanks for linking that issue though!

nouser2013, 2019-08-23:
I'll chime in here, as this problem started to occur a few days ago on my installation as well, on a DeltiQ XL running a custom port of Marlin 1.0.1.

- No changes in hardware (Same SD, USB cable, PI, RAMPS board, Firmware, Printer)
- No changes on the OctoPrint (albeit some autoupdates enabled? it does have internet access)
- Same GCode printed flawlessly last week.
- This week: blobs visible on the outside, 50-100 1-3 second stops during a 7 hours print, reproducably with all prints so far.
- This week: tried to print in safe mode: only one print finished with only 1 blob. Second print running at the moment, will update later today.

I have only updated the external LCD printer display plugin after a release to 1.13 was advertised on the OctoPrint front page, and shortly after that to 1.14. Only two other plugins are and were active throughout the whole life of the installation.

Other than that, I have no idea what triggered this, and would like to follow this (possible) bug here. I have not tried to print from SD card yet, the FW does neither recognize `*.gco` nor `*.gcode` files. The SD-Card menu on the printer is empty.

A fix I have read somewhere is to force the USB mode to USB 1.1 on the Pi OS, but I have not tried that yet, also.

nouser2013, 2019-08-23:
Update: The second 7-hours print has finished in safe mode. There was only one print quality issue in it and no visible blobs. However, I'm seeing a lot of these in the `octoprint.log`:

```
2019-08-22 23:36:17,458 - Send: N96180 G1 X42.846 Y21.662 E0.05385*98
2019-08-22 23:36:17,461 - Send: N96181 G1 X42.292 Y21.108 E0.04042*98
2019-08-22 23:36:17,469 - Recv: ok 96178
2019-08-22 23:36:17,471 - Recv: ok 96179
2019-08-22 23:36:17,474 - Send: N96182 G1 X42.742 Y20.154 E0.05441*103
2019-08-22 23:36:17,475 - Send: N96183 G1 X43.303 Y20.715 E0.04091*109
2019-08-22 23:36:17,476 - Recv: ok 96180
2019-08-22 23:36:17,478 - Send: N96184 G1 X43.734 Y19.792 E0.05256*103
2019-08-22 23:36:17,484 - Recv: ok 96181
2019-08-22 23:36:17,486 - Recv: ok 96182
2019-08-22 23:36:17,488 - Send: N96185 G1 X43.748 Y19.757 E0.00196*110
2019-08-22 23:36:17,491 - Send: N96186 G1 X43.180 Y19.188 E0.04146*98
2019-08-22 23:36:17,517 - Recv: ok 96183
2019-08-22 23:36:17,520 - Recv: ok 96184
2019-08-22 23:36:17,521 - Send: N96187 G1 X43.599 Y18.203 E0.05523*104
2019-08-22 23:36:17,523 - Send: N96188 G1 X44.167 Y18.771 E0.04146*99
2019-08-22 23:36:17,549 - Recv: ok 96185
2019-08-22 23:36:17,551 - Send: N96189 G1 X44.386 Y18.253 E0.02900*102
2019-08-22 23:36:17,594 - Recv: Error:Wrong checksum
2019-08-22 23:36:17,595 - Recv: 
2019-08-22 23:36:17,596 - Recv: Resend:96186
2019-08-22 23:36:17,600 - Recv: ok
2019-08-22 23:36:17,601 - Recv: skip 96188
2019-08-22 23:36:17,602 - Send: N96186 G1 X43.180 Y19.188 E0.04146*98
2019-08-22 23:36:17,604 - Recv: ok
2019-08-22 23:36:17,605 - Send: N96187 G1 X43.599 Y18.203 E0.05523*104
2019-08-22 23:36:17,606 - Recv: skip 96189
2019-08-22 23:36:17,607 - Recv: ok
2019-08-22 23:36:17,609 - Send: N96188 G1 X44.167 Y18.771 E0.04146*99
2019-08-22 23:36:17,609 - Recv: ok 96186
2019-08-22 23:36:17,611 - Send: N96189 G1 X44.386 Y18.253 E0.02900*102
2019-08-22 23:36:17,625 - Recv: ok 96187
2019-08-22 23:36:17,628 - Send: N96190 G1 X44.571 Y17.772 E0.02662*98
2019-08-22 23:36:17,642 - Recv: ok 96188
2019-08-22 23:36:17,644 - Send: N96191 G1 X43.995 Y17.195 E0.04208*99
2019-08-22 23:36:17,657 - Recv: ok 96189
2019-08-22 23:36:17,659 - Send: N96192 G1 X44.376 Y16.173 E0.05629*111
2019-08-22 23:36:17,690 - Recv: ok 96190
2019-08-22 23:36:17,692 - Send: N96193 G1 X44.961 Y16.757 E0.04263*105
2019-08-22 23:36:17,721 - Recv: ok 96191
2019-08-22 23:36:17,724 - Send: N96194 G1 X44.985 Y16.693 E0.00352*106
2019-08-22 23:36:17,754 - Recv: ok 96192
2019-08-22 23:36:17,756 - Send: N96195 G1 X45.322 Y15.715 E0.05340*103
2019-08-22 23:36:17,785 - Recv: ok 96193
2019-08-22 23:36:17,788 - Send: N96196 G1 X44.736 Y15.129 E0.04277*105
2019-08-22 23:36:17,817 - Recv: ok 96194
2019-08-22 23:36:17,820 - Send: N96197 G1 X45.072 Y14.061 E0.05776*103
```

But apparently, these errors do not affect print quality. Also, I cannot say if this is a new problem, never had the logging enabled prior to a few days ago.

FormerLurker, 2019-08-23:
So, I have NOT been able to replicate my original problem recently.  I have tried and tried, so I'm chalking it up to some kind of fluke.  I'm thinking that I enabled linear advance by mistake by changing profiles, and I've seen that cause just this issue in the past.  @nouser2013, I'm going to close this issue unless you object, so you might want to create a new one if you are still having issues.  I don't want to pollute the issues with something that I can't even verify.

closed 2019-08-23 by FormerLurker