OctoPrint/OctoPrint #3236: Upload folder not being read - exactly the same problems as #3022 but different trigger

gcw843, 2019-07-24:
Octoprint 1.3.11 running on Octopi .16 on RPI 3b+ with wi-fi connection to lan experiencing exactly the same problem as #3022 due to Cura 4.1. 

I upgraded from Cura 3.6 to Cura 4.1 and checked the box in Cura's machine settings to connect Cura to Octoprint. 

I later decided to disconnect Octoprint from Cura and unchecked the box. I then experienced behavior identical to issue #3022: Upload folder not being read:

The saving bar in the files section hung and never completed. 

Gcode files were uploading to the correct folder on the pi but were not being displayed in Octoprint's file area.  

The print time showed naN and z height was infinity in /uploads/metadata.json.  

I have never downloaded calibration gcode for lin_advance calibration. 

Deleting the Gcode file and the metadata file would allow another gcode file with a different name to be uploaded and shown in the files section but upload & print wouldn't work.

Running in safe mode made no difference.  

I reinstalled Octopi several times and each time updated to Octoprint 1.3.11. The problem persisted across 4-5 reinstallations.

There were no new plugins or hardware changes. The only change was the upgrade to Cura 4.1 and the checking and then unchecking the box to connect Cura to Octoprint in Cura's machine settings.

I finally uninstalled Cura 4.1 and reinstalled Cura 3.6 and it solved the problem. Octoprint is back to working beautifully.

I've enjoyed your excellent program since the early days and wanted to say thank you for creating it Gina.  

This isn't a request for assistance - just wanted to give you a heads up about the issue and to let you know your work is appreciated! 

Gene

foosel, 2019-07-25:
@fieldOfView Can you shed any light on possible differences between the integration in Cura 3 and Cura 4 that could be contributing to this?

fieldOfView, 2019-07-25:
I am a bit confused; there is no box to check or uncheck in Machine Settings to connect to OctoPrint. As a matter of fact, there’s no real way to sever the connection between a printer in Cura and an OctoPrint instance (I have been meaning to add it somehow, but it seems noone has missed it sofar)

foosel, 2019-07-25:
@gcw843 could you clarify with some screenshots maybe?

gcw843, 2019-07-25:
No. I wish I'd thought to grab some while I was having the problem. It started several weeks ago so I'm relying on memory unfortunately. I uninstalled Cura 4.1 which solved the issue that had persisted for a couple of weeks through several complete reinstalls of Octopi & Octoprint. In Cura 4.1 it was in machine settings (manage printers? maybe) and the check box was either connect to octoprint or discover octoprint or something like that. While I was trying to figure it out I didn't think to document the settings.

fieldOfView, 2019-07-26:
That clears it up a little. You are talking about the checkbox I added to disable the (pure-Python) Zeroconf/bonjour autodiscovery. 

I think the change we are talking about is not so much about Cura, but the plugin. To fix some issues with the autodiscovery (eg for names with dots in them), I started using a newer version of the python zeroconf module. Apparently, that has caused other issues on some peoples networks. Since then I have done some more work on the zeroconf version included, and not knowing what version of the OctoPrint Connection plugin you were using I don't know if that solved the issue or not.

Note that even without the autodiscovery, you can still use the plugin. You would have to know the ip address or the hostname of the octoprint instance though.

foosel, 2019-07-26:
Sounds like there might be some transfer issue? In any case, 4cb67fe72c5e248d3a6e5d46a9909463d4dd8866 as pushed for #3022 should prevent `NaN` from ending up in the metadata, so with 1.3.12 (aka what's currently on the `maintenance` branch) this should no longer happen.

gcw843, 2019-07-26:
@foosel: I saw in #3022 that you had already figured out a fix for the upload folder issue. Just wanted to give you a heads up about the Cura issue.

@fieldofview: I don't know what version of the plugin I was using and I've already uninstalled it.

foosel, 2019-07-26:
Closing this as a duplicate then as it's simply a different set of steps to the same issue and there's (probably) nothing to be done here anymore.

closed 2019-07-26 by foosel