OctoPrint/OctoPrint #3237: "After print job is paused" gcode does not run sometimes on pausing.

vadsh, 2019-07-27:
#### What were you doing?
Pause a print via rest api or filament runout plugin (GPIO). As far as I know pause button in web-ui have the same problem.
When I pause a print via filament runout plugin (GPIO) or via REST API usually (about 80% of times) everything is ok and after pausing appropriate g-code from "After print job is paused" section in settings runs.

#### What did you expect to happen?
Print pauses and "After print job is paused" gcode runs.
Printer LCD shows paused state and I can manipulate with movement (marlin menu: motion -> move axis)

#### What happened instead?
Usually all works as expected. But once in a while (about 15% of times) printer just pauses without "After print job is paused" does not run.
State of printer stays "Printing" and I can not manipulate with movement (as printer thinks that job is in progress)

Octoprint in this case still remembers the print position and I can resume the printing.
Filament change is very troublesome in this case as "After print job is paused" gcode was not run.

Sometimes printer prints for quite a while after receiving pause command (due to large buffers on marlin I suppose).
Here are marlin buffer section
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 64 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 64 // maximize block buffer
#endif

// @section serial

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 65 
#define BUFSIZE 32

// Transmission to Host Buffer Size
// To save 386 bytes of PROGMEM (and TX_BUFFER_SIZE+3 bytes of RAM) set to 0.
// To buffer a simple "ok" you need 4 bytes.
// For ADVANCED_OK (M105) you need 32 bytes.
// For debug-echo: 128 bytes for the optimal speed.
// Other output doesn't need to be that speedy.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256]
#define TX_BUFFER_SIZE 128

// Host Receive Buffer Size
// Without XON/XOFF flow control (see SERIAL_XON_XOFF below) 32 bytes should be enough.
// To use flow control, set this buffer size to at least 1024 bytes.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048]
#define RX_BUFFER_SIZE 128

#### Did the same happen when running OctoPrint in safe mode?
Unknown

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
octopi 4.19.42-v7+ 

#### Printer model & used firmware incl. version
Marlin 2.0.x bugfix. SKR V1.3 (the same happened with marlin 1.9 and several 8-bit boards)

#### Browser and version of browser, operating system running browser
Browser is not in action here. 

#### Link to octoprint.log
[octoprint (5).log](https://github.com/foosel/OctoPrint/files/3437672/octoprint.5.log)

#### Link to contents of terminal tab or serial.log
not available

#### Link to contents of Javascript console in the browser
not available

#### Screenshot(s)/video(s) showing the problem:
na

foosel, 2019-07-29:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

vadsh, 2019-07-30:
@foosel , thanks for reply.

I've increased all timeouts at "Settings" -> "Serial Connection" -> "Query intervals" page

The problem seems to be solved for now.

Is that possible that problem was due to large marlin buffers (sometimes printer continues to print for about a minute after the pause button is pressed) and low default timeouts?

foosel, 2019-08-06:
> Is that possible that problem was due to large marlin buffers (sometimes printer continues to print for about a minute after the pause button is pressed) and low default timeouts?

Maybe, but sadly impossible to say without a `serial.log`. Since apparently it is solved for you now, any objections to closing this?

closed 2019-08-07 by vadsh