OctoPrint/OctoPrint #3238: [Request] PyPi release

halkeye, 2019-07-27:
**Is your feature request related to a problem? Please describe.**
I would like to make my own docker image that is always up to date. My plan for this was to use https://dependabot.com/ to create a new build every time a new package or plugin was updated using requirements.txt

I quickly ran into a snag because octoprint itself is not on pypi

**Describe the solution you'd like**
Release in parallel to pypi

**Describe alternatives you've considered**
I can work around it by assuming plugins will eventually update, manually building, or doing timed builds.

I'm hoping dependabot has support for github urls, or using github package releases when the feature is out of alpha.

I tried doing a quick search in past issues and don't see it brought up, so I thought I'd ask

foosel, 2019-07-29:
That's actually something I've been meaning to do for ages now... I really need to finally get around to it.

rooterkyberian, 2019-07-30:
anything we can do to help?

foosel, 2019-09-04:
1.3.11 should now be available through a simple `pip install OctoPrint`. I'll look into setting up a Github Action or something like this to automate that on future releases.

closed 2019-09-04 by foosel

halkeye, 2019-09-05:
Awesome. My docker file works now. Easy to keep up to date now!

Thanks so much