OctoPrint/OctoPrint #3241: [Request] Configurable NTP server

pompel82, 2019-07-31:
The problem: Booting OctoPrint takes a very long time as it's behind a restrictive firewall and all NTP attempts are failing. There seems to be lot's of non-standard NTP servers configured by default. My firewall will not allow connecting to any of these servers so all will time-out.

My preferred solution: Add an "NTP server" config item so I can configure my local gateway as the NTP server to be used.

Alternative: Use a well known and trusted NTP server such as e.g. time.nist.gov

Additional context: NA

ManuelMcLure, 2019-07-31:
This is more an OctoPi issue than an OctoPrint one - it would be better to enter this issue at https://github.com/guysoft/OctoPi

foosel, 2019-08-01:
What @ManuelMcLure said - wrong project for this request.

closed 2019-08-01 by foosel

pompel82, 2019-08-01:
Turned out to be a Raspbian issue and solved it be changing timesyncd.conf as described here: [https://raspberrypi.stackexchange.com/questions/85670/how-to-set-custom-ntp-servers-in-recent-raspbian-stretch-systems](https://raspberrypi.stackexchange.com/questions/85670/how-to-set-custom-ntp-servers-in-recent-raspbian-stretch-systems)