OctoPrint/OctoPrint #3242: [FEATURE] Add stop button to the control menu.

k0nradz, 2019-08-01:
Hi,

It is possible to add STOP button to the control menu? When we push home button manually its im possible to stop the printer. Today I have collision nozzle with bed after manually push home button.

[image]

foosel, 2019-08-01:
Take a look at one of the existing emergency stop plugins and custom controls.

Apart from that it's the job of your printer's firmware to keep the nozzle from ramming into the print bed, especially while homing.

closed 2019-08-01 by foosel

k0nradz, 2019-08-01:
I have found "Custom Control Editor (0.2.1)" plugin. It works great when I added "M84;" G-Code stop command. 

THANKS FOR HELP!