OctoPrint/OctoPrint #3250: [Request] Don't allow a print to be started by keyboard only/github like confirmation

dadosch, 2019-08-14:
**Is your feature request related to a problem? Please describe.**
The following situation: a person who doesn't know 3d printing (for example family member) tries to log in to a computer. To wake up the screen she presses enter, but since the screen isn't locked (just turned off), octoprint starts to print since the tab hasn't been close and fokus was on the print button. The printer then starts to print and might damage itself or the previous print. This might sound made up, but it just happened to me.

**Describe the solution you'd like**
A solution to this problem could be to allow a print only after a confirmation dialog is clicked (which can't be bypassed by accident) or a github like confirmation, where you have to type for example the first word of the stl file/your username/whatever.

**Describe alternatives you've considered**
An alternative would be to timeout octoprint sessions.

skylu3d, 2019-08-23:
Hi, 
I just having some issue with have the octoprint to read and printing from yesterday.
I was able to load the gcode file and but when I click print from browser and it won't process the gcode...
so I restart octopi in safe mode and try again but seems it works fine.
Is there something can be done to resolve this issue?

Thanks in advanced.
[image]
[image]

dadosch, 2019-08-23:
> Hi,
> I just having some issue with have the octoprint to read and printing from yesterday.
> I was able to load the gcode file and but when I click print from browser and it won't process the gcode... [...]

I think this isn't related to this request. Would you mind using the forum instead:

https://community.octoprint.org/

foosel, 2019-09-05:
There's now an optional print start confirmation:

[image]

Check that box and you'll get a sanity check confirmation dialog when starting a print job via the UI:

[image]

Pushed to `maintenance`, soon `devel`, to be part of 1.3.12.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel