OctoPrint/OctoPrint #3251: printjob cancelled right after starting

Leprechaun91, 2019-08-16:
#### What did you expect to happen?
Printer should start printing
#### What happened instead?
Printing is started and heating starts, seconds later status changed to "cancelling"
#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
Raspi

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Leprechaun91, 2019-08-16:
`

2019-08-16 14:01:39,023 - octoprint.plugins.telegram - DEBUG - Saving data: {u'chats': {u'382113096': {u'commands': {u'/filament': True, u'/upload': True, u'/con': True, u'/print': True, u'/gif': True, u'/status': True, u'/user': True, u'/abort': True, u'/dontshutup': True, u'/shutup': True, u'/files': True, u'/supergif': True, u'/ctrl': True, u'/togglepause': True, u'/tune': True, u'/settings': True, u'/sys': True}, u'title': u'Julian - Kusenberg - ', u'image': u'/plugin/telegram/img/user/pic382113096.jpg', u'send_notifications': True, u'private': True, u'allow_users': False, u'notifications': {u'StatusPrinting': False, u'PrintDone': True, u'ZChange': True, u'PrinterStart': True, u'PrintFailed': True, u'StatusNotPrinting': False, u'PrintStarted': True, u'PrintResumed': False, u'PrintPaused': True, u'PrinterShutdown': True}, u'newUsr': False, u'new': False, u'type': u'PRIVATE', u'id': u'382113096', u'accept_commands': True}}}
2019-08-16 14:01:45,594 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client ::1
2019-08-16 14:01:51,688 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-08-16 14:01:53,331 - octoprint.server.preemptive_cache - INFO - ... done in 14.39s
2019-08-16 14:02:01,614 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2019-08-16 14:02:01,816 - tornado.access - WARNING - 409 GET /api/printer?exclude=sd,history (::ffff:192.168.2.63) 11.85ms
2019-08-16 14:02:22,342 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-08-16 14:02:31,863 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2019-08-16 14:02:52,393 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-08-16 14:03:01,876 - tornado.access - WARNING - 409 GET /api/printer?exclude=sd,history (::ffff:192.168.2.63) 23.07ms
2019-08-16 14:03:02,103 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2019-08-16 14:03:22,168 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2019-08-16 14:03:22,174 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2019-08-16 14:03:22,182 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-08-16 14:03:22,441 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-08-16 14:03:23,804 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-08-16 14:03:32,020 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2019-08-16 14:03:32,048 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-08-16 14:03:32,105 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin SkyNet3D-V2.3.2 (Github)"
2019-08-16 14:03:32,312 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2019-08-16 14:03:49,534 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Back.gcode, owner: admin, user: admin
2019-08-16 14:03:49,546 - octoprint.plugins.DisplayLayerProgress - INFO - File selected. Determining number of layers.
2019-08-16 14:03:52,125 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2019-08-16 14:03:52,137 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Back.gcode, owner: admin, user: admin
2019-08-16 14:03:52,522 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-08-16 14:03:52,771 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2019-08-16 14:03:52,963 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-08-16 14:03:53,167 - octoprint.plugins.cancelobject - INFO - Tracking Extrusion
2019-08-16 14:04:02,576 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2019-08-16 14:04:08,919 - octoprint.plugins.DisplayLayerProgress - INFO - Printing started. Detailed progress started.{'origin': 'local', 'name': u'Back.gcode', 'file': u'/home/pi/.octoprint/uploads/Back.gcode', 'owner': 'admin', 'path': u'Back.gcode', 'user': 'admin', 'size': 22129517L, 'filename': u'Back.gcode'}
2019-08-16 14:04:08,939 - octoprint.plugins.IFTTT - INFO - .name
2019-08-16 14:04:08,940 - octoprint.plugins.IFTTT - INFO - .origin
2019-08-16 14:04:08,943 - octoprint.plugins.IFTTT - INFO - 
2019-08-16 14:04:13,253 - octoprint.plugins.filamentreload - INFO - Printing aborted: no filament detected!
2019-08-16 14:04:13,264 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2019-08-16 14:04:13,300 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Back.gcode, owner: admin, user: None
2019-08-16 14:04:13,355 - octoprint.plugins.filamentreload - INFO - PrintStarted: Enabling filament sensor.
2019-08-16 14:04:13,781 - octoprint.printer.standard - ERROR - Exception while pushing current data to callback <layerdisplay.LayerDisplayPlugin object at 0x72ca4530>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 196, in _sendCurrentDataCallbacks
    callback.on_printer_send_current_data(copy.deepcopy(data))
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 437, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 412, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2019-08-16 14:07:54,053 - tornado.general - ERROR - Cannot send error response after headers written
``

tedder, 2019-08-16:
Run in safe mode and upload that log.

bastilian, 2019-08-31:
These lines of the log might be an indicator what is happening:
```
2019-08-16 14:04:08,943 - octoprint.plugins.IFTTT - INFO -
2019-08-16 14:04:13,253 - octoprint.plugins.filamentreload - INFO - Printing aborted: no filament detected!
2019-08-16 14:04:13,264 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
```

If you have a filament sensor setup, ensure it is pluggged in correctly, configured correctly and working as it should.

Additionally it appears the telegram plugin fails to send a notification, which is the greater part of the rest of the log.

foosel, 2019-09-04:
As already pointed out, that looks like its caused by your third party filament detector plugin detecting a lack of filament and cancelling the print.

closed 2019-09-04 by foosel