OctoPrint/OctoPrint #3253: Error Detecting Python Version - Plugin Development

johnnyruz, 2019-08-21:
#### What were you doing?

1. Follow instructions on the Plugin Development Tutorial page to setup the OctoPrint Development environment
2. Once successfully installed, execute command `octoprint serve`
3. View the output of the server startup log
4. See error
```
2019-08-21 11:55:47,421 - octoprint.plugin - ERROR - Error while calling plugin softwareupdate
Traceback (most recent call last):
  File "/home/multipass/OctoPrint/src/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/multipass/OctoPrint/src/octoprint/plugins/softwareupdate/__init__.py", line 102, in on_after_startup
    self._check_environment()
  File "/home/multipass/OctoPrint/src/octoprint/plugins/softwareupdate/__init__.py", line 184, in _check_environment
    versions = dict(python=get_python_version_string(),
  File "/home/multipass/OctoPrint/src/octoprint/util/version.py", line 126, in get_python_version_string
    version_string = python_version[:-1]
TypeError: 'function' object has no attribute '__getitem__'
```

#### Proposed Solution
Update line the get_python_version_string() method (line 126) in file src/octoprint/util/version.py from

`version_string = python_version[:-1]`

to

`version_string = version_string[:-1]`

#### What did you expect to happen?
OctoPrint server should start successfully without error

#### What happened instead?
Received the error above

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
Ubuntu 18.04LTS

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
N/A

johnnyruz, 2019-08-21:
See community thread: https://community.octoprint.org/t/error-detecting-python-version/10669

foosel, 2019-09-04:
Fixed by #3256 - thanks! Will be in 1.3.12.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel

johnnyruz, 2019-10-24:
Confirmed this is resolved.