OctoPrint/OctoPrint #3261: Marks on the temperature graph [Request]

quizwammer, 2019-09-02:
It would be nice to tell how the temperature relates with pausing/stopping prints, so marks/lines indicating when the print was started/paused/stopped etc would be useful. This would be done by waiting for specific gcode commands, and logging them with the time to display them on the temperature graph.

Thanks, Quentin

surdu, 2022-01-03:
[image]

@foosel @qduff  Will this suffice ?

cp2004, 2022-01-03:
I would like some kind of key showing what they mean - but otherwise, I think something like this could work. I guess green is start and red is stop...

surdu, 2022-01-04:
@cp2004 I've updated the above screenshot to also contain labels

surdu, 2022-01-08:
@cp2004 @foosel Quick question: will the marking need to survive a reload? (add the data to the historyData maybe ?)

foosel, 2022-01-17:
IMHO the data should definitely survive a reload and thus needs to be persisted server side. If we just add it to the graph as it happens and a reload makes it go away, I would imagine that would greatly confuse people and lead to the one or other ticket as well.