OctoPrint/OctoPrint #3262: Bad character (Ascii 150) in plugin can 500 web interface.

Kragrathea, 2019-09-04:
I was doing some plugin development and I came across a character IN A COMMENT that will cause the web interface to return a 500 internal server error.  Rebooting would not fix, I had to track down the character and remove it before I could do anything in Octoprint. 

The comment was:
                    //G0/G1 – Linear Movement

And the problem character is the –. Its is not a usual minus sign - (Ascii 45) it is a dash (Ascii 150) and as far as I can tell if it is anywhere in a plugin .JS the web interface will crash. Seems to be in flask somewhere.

foosel, 2019-09-05:
Can you provide a test JS file and most importantly a log? I can't reproduce this.

foosel, 2019-09-16:
@Kragrathea ping?

Kragrathea, 2019-09-17:
Sorry, I just saw your reply. I retested by copying this string into my plugins main .js file.

-I deleted my logs
-Restarted OctoPrint and verified it worked
-Pasted this into my .JS file.

//G0/G1 – Linear Movement

I got the same 500 error as before. My OctoPrint is 1.3.11 running on a regular Pi with Wifi (I forget the model but not Zero). 

Here is the relevant part of the log.
2019-09-17 06:00:12,361 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.7) 14.66ms
2019-09-17 06:00:12,469 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.7) 10.57ms
2019-09-17 06:00:22,524 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 422, in index
    response = default_view()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 390, in default_view
    return preemptively_cached()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 984, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1047, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1055, in decorated_function
    r = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 747, in decorated_function
    rv = f_with_duration(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 716, in f_with_duration
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 993, in decorated_function
    rv = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1012, in decorated_function
    rv = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 377, in make_default_ui
    r = make_response(render_template("index.jinja2", **render_kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/templating.py", line 128, in render_template
    context, ctx.app)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/templating.py", line 110, in _render
    rv = template.render(context)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/jinja2/environment.py", line 989, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/jinja2/environment.py", line 754, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/templates/index.jinja2", line 144, in top-level template code
    {% include 'javascripts.jinja2' %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/templates/javascripts.jinja2", line 3, in top-level template code
    {% assets "js_plugins" %}<script type="text/javascript" src="{{ ASSET_URL }}"></script>{% endassets %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/ext/jinja2.py", line 186, in _render_assets
    urls = bundle.urls()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 806, in urls
    urls.extend(bundle._urls(new_ctx, extra_filters, *args, **kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 765, in _urls
    *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 619, in _build
    force, disable_cache=disable_cache, extra_filters=extra_filters)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 504, in _merge_and_apply
    filters_to_pass_down, disable_cache=disable_cache)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 148, in _merge_and_apply
    disable_cache=disable_cache)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 543, in _merge_and_apply
    kwargs=item_data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/merge.py", line 276, in apply
    return self._wrap_cache(key, func)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 220, in fixed_wrap_cache
    content = self.cache.get(key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 183, in fixed_get
    hash = make_md5(self.V, key)
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/merge.py", line 79, in data
    return f.read()
  File "/home/pi/oprint/lib/python2.7/codecs.py", line 314, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x96 in position 161: invalid start byte
2019-09-17 06:00:22,528 - tornado.access - ERROR - 500 GET / (::ffff:192.168.1.7) 6110.52ms
2019-09-17 06:00:22,560 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.7) 9.88ms

Kragrathea, 2019-09-17:
Here is a minimal JS file that crashes mine. 

`$(function () {
    console.log("Create PrettyGCode View Model");
    function PrettyGCodeViewModel(parameters) {

//THE COMMENT BELOW CAUSES THE CRASH.
//G0/G1 – Linear Movement

    }

    // This is how our plugin registers itself with the application, by adding some configuration
    // information to the global variable OCTOPRINT_VIEWMODELS
    OCTOPRINT_VIEWMODELS.push([
        // This is the constructor to call for instantiating the plugin
        PrettyGCodeViewModel,

        // This is a list of dependencies to inject into the plugin, the order which you request
        // here is the order in which the dependencies will be injected into your view model upon
        // instantiation via the parameters argument
        ["settingsViewModel"],

        // Finally, this is the list of selectors for all elements we want this view model to be bound to.
        ["#injector_link"]
    ]);
});`

foosel, 2019-09-18:
I still cannot reproduce this. I tried against 1.3.11 and what's currently on `maintenance`, and earlier also against what is now 1.3.12rc2. No issues at all.

Here's the repository with my test plugin: https://github.com/foosel/OctoPrint-Issue-3262.

foosel, 2019-09-18:
For good measure I now also tried as a full plugin package installed through `pip`. Cannot reproduce.

Kragrathea, 2019-09-19:
Then it must be something about my setup. I do have a few other plugins installed. Touch Bed UI and Layer number in the Status bar. 

But I just worked around it by removing the bad character. So I don't need a fix and am happy if you want to close this as 'can't repro'. If I do find any more info I will update here.

foosel, 2019-09-19:
I'll set it to unreproduced for now, maybe someone else will stumble over it as well.

DanielJoyce, 2019-09-26:
```
File "/home/pi/oprint/lib/python2.7/codecs.py", line 314, in decode
(result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x96 in position 161: invalid start byte
```

Its an encoding issue. Kragrathea likely has his  filesystem using a non-utf8 encoding, Python process is running as utf-8 though, and tries to load the file as utf-8, and barfs on the character.

Ascii code 150 is invalid in utf-8, it would be encoded as two bytes.

Check the file system encoding.

foosel, 2019-11-05:
Closing as still unreproduced and most likely an environment issue.

closed 2019-11-05 by foosel