OctoPrint/OctoPrint #3264: Python 2 Deprecation Plan?

lovesegfault, 2019-09-05:
Since the [deprecation of Python 2 is looming](https://pythonclock.org/), I wanted to ask what is OctoPrint's action plan in this regard? Is it going to continue with Python 2 even after it's unmaintained, or will is there a transition plan?

foosel, 2019-09-05:
In a nutshell:

  * 1.4.0 will support Python 3 and 2 to allow users & plugin authors to migrate (see the `devel` branch) - to be released this year (somehow... aaaaaaah)
  * 1.5.0 aka 2.0.0 will only support Python 3

There are over 50000 confirmed instances of OctoPrint out in the wild on various base images, simply dropping support for Python 2 on January 1st is therefore not an option, plus there's a whole ecosystem of plugins to think about as well.

I've also addressed this in various episodes of [OctoPrint On Air](https://www.youtube.com/playlist?list=PL9j2DtsIPVkOFIMRrnnbXsnXtQmwj1IId). See also [this forum thread](https://community.octoprint.org/t/octoprint-support-for-python-2-3-versus-the-plugin-space/7753?u=foosel).

closed 2019-09-05 by foosel

lovesegfault, 2019-09-05:
@foosel Thank you!