OctoPrint/OctoPrint #3265: No Notification Messages on Webinterface & Slow Webinterface

xenon2008, 2019-09-08:
#### What were you doing?
Hello!

Since i Yesterday have Updated the Plugin "PrinterStats" to Version 2.0, my Octopi Installations are very Slow and i also get no Notifications on Webinterface any more.
Like by failed or Successfully Prints etc...
Or if a Timelapse is rendering.... no more Messages....

I also noticed that my Octoprint is now extremely slow!
The installation of the new version of the Printerstats plugin lasted all night!

I hope everybody can understand my bad English. I am a German User with terrible English Skills! ;)

I am using Octopi

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Updated the Plugin "PrinterStats" to Version 2.0

First i also get an Error like described here:
https://github.com/amsbr/OctoPrint-Stats/issues/18

Then i have run the Update again, and let it Update over Night!
The Update was successfull, but since this i have no Notification Messages any more and my Octoprint Installations is very slow!

#### What happened instead?
N/A
#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
OctoPrint 1.3.11 auf OctoPi 0.16.0 on RPi3b

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version
Ender 3 Pro with SKR 1.3 and Marlin 2.0 Bugfix

#### Browser and version of browser, operating system running browser
Google Chrome (latest) & Mozilla FF (latest)

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/3587588/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2019-09-08:
Did the same happen when running OctoPrint in safe mode?

xenon2008, 2019-09-08:
not have tried!
but i have disabled the PrinterStats Plugin, and it seems to work.
The Update of this Plugin also installed " numpy", maybe this is the problem?

I already have tried to reinstall "PrinterStats" without Success...

tedder, 2019-09-08:
Run in safe mode. You have several plugins and at least two are failing- filamentmanager and telegram.

xenon2008, 2019-09-08:
okay i will try this as soon as i can.
But since i have disabled "Printerstats" Filamentmanager & telegram also works , and i also get notifications when a timelapse is rendering ...

amsbr, 2019-09-09:
@xenon2008 all operations at UI interface are slow or only notifications?

When you should have notifications and they are not displayed, does the PrintStats progress icon appear?

xenon2008, 2019-09-09:
@amsbr 
all operations at UI are slowly. And sometimes the prints stucks / stopps for seconds....

And when PrinterStats is disabeld i cant print without any Problems.
And then i also get immediately Notifcations if a Print is failed, oder a timelapse is rendering at the moment.

it seems like the plugin is generating a high load.
And that it really is this numpy?

foosel, 2019-09-09:
> And when PrinterStats is disabeld i cant print without any Problems.
> And then i also get immediately Notifcations if a Print is failed, oder a timelapse is rendering at the moment.

Then your problem lies within PrinterStats, which is a third party plugin not under my control, not OctoPrint. Which safe mode would also have shown you. Closing this since it's out of my control.

closed 2019-09-09 by foosel