OctoPrint/OctoPrint #3266: System clock out of sync on boot

seltix5, 2019-09-08:
#### What were you doing?

Already tested this on a new instalation with no plugins installed.

1. Cut power off from raspberry pi ( without using the Shutdown system from UI )
2. wait something like 5 or 10min
3. Boot up Octoprint
4. Check octoprint.log, the timestamp will be equal or INFERIOR at timestamp before shutdown

#### What did you expect to happen?

I wold expect the clock to be updated before loading plugins because. The wrong datetime, specially when the timestamp is inferior at the shutdown timestamp, it will cause some security errors with some addons using tokens and access timestamp validation.

#### What happened instead?

system clock is only updated after plugins are loaded.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.11

#### Operating System running OctoPrint

OctoPi 0.16.0

#### Printer model & used firmware incl. version
[octoprint.log](https://github.com/foosel/OctoPrint/files/3587867/octoprint.log)

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

for example :
octoprint boot up at :
`2019-09-08 18:50:45,224 - octoprint.startup - INFO - Starting OctoPrint 1.3.11`
almost at the same time as previous log (2019-09-08 18:50:34,382) yet almost 10 min have passed from when it was shutdown...

I have read the FAQ.

tedder, 2019-09-09:
The Raspberry Pi doesn't have a battery to keep time while it is turned off. There are [little battery clocks you can add](https://www.google.com/search?q=raspberry+pi+rtc), presumably most people leave it running so they don't notice. You can [debug why it takes a while to catch up](https://raspberrypi.stackexchange.com/questions/8231/how-to-force-ntpd-to-update-date-time-after-boot) and/or add a sync to rc.local on start. Not a Octoprint issue though.

seltix5, 2019-09-09:
Hello, thank you, I will try to debug it then and report back, it may help others.
Thanks

seltix5, 2019-09-09:
Hello, Ok I was reading about it and I understand why the clock does not has the right time on boot and I can't install something like a cheap Ds3231 clock module with battery to keep the correct time for when Pi has no power. What I don't get is, if it saves the last know timestamp witch is great to keet a relative accurate date, why does it go backwards some times?

foosel, 2019-09-09:
As has already been pointed out, that's an issue with the underlying platform, not OctoPrint running on it.

closed 2019-09-09 by foosel

seltix5, 2019-09-09:
issue created in Octopi : https://github.com/guysoft/OctoPi/issues/606