OctoPrint/OctoPrint #3267: Octoprint not autoconnecting on startup if port is set to AUTO

seltix5, 2019-09-08:
#### What were you doing?

1. Cut power off from raspberry pi ( without using the Shutdown system from UI )
2. wait something like 5 or 10min
3. Boot up Octoprint
4. Check Octoprint UI > Connection

#### What did you expect to happen?

Octoprint should connect to the printer on boot with the port set to AUTO.

#### What happened instead?

Did not connect.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

OctoPrint 1.3.11

#### Operating System running OctoPrint

OctoPi 0.16.0

#### Printer model & used firmware incl. version

Ender 3 with original Marlin 1.1.6

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/3587900/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-09-11:
>     1. Cut power off from raspberry pi ( without using the Shutdown system from UI )

Please never ever do this. It can and with a bit of bad luck will destroy data, often in subtle ways that cause weird issues that look like bugs.

Still will look into this ticket though.

seltix5, 2019-09-11:
hello,
even when is doing nothing? because octoprint uses the printer power supply when I shut it down the PI goes with it and some times I do not shutdown correctly just because of the hassle of going to the pc just to power off. I will be more careful in the future.

foosel, 2019-09-11:
> even when is doing nothing?

Yes, even when it's "doing nothing" (it might seem to not be doing anything, but that doesn't mean it's not writing to disk in the background). Never just shut down a Pi. You risk data corruption and loss.

seltix5, 2019-09-11:
thanks, I will be more careful in the future. and I will search if I can add a physical button to shutdown.

I was doing a quick test, I just did 2 Reboot system in a row and it did not connect to the printer (in AUTO)

foosel, 2019-09-11:
I might have to add some more debugging statements in order to figure this out, would you be comfortable running a custom branch?

seltix5, 2019-09-11:
Sure, no problem

foosel, 2019-09-11:
Could you enable your `serial.log` (it's currently disabled), reproduce the error and then share the log? Looking at the code it should try to auto connect, but it could be that the port is not yet registered in the system (as in, OctoPrint boots faster than your printer) in which case it can't possibly auto connect right on boot up.

Also, I just spotted this in your `octoprint.log`:

```
2019-09-08 18:34:47,607 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
[...]
2019-09-08 18:34:47,636 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
[...]
2019-09-08 18:34:47,642 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2019-09-08 18:34:47,648 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
[...]
2019-09-08 18:34:48,024 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
[...]
2019-09-08 18:34:50,962 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2019-09-08 18:34:50,971 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-09-08 18:34:51,044 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.6 (Github)"
2019-09-08 18:34:51,070 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
```

That very much looks like an auto connect to with port detection to me, and a successful one to boot.

But for some reason it got disconnected half a minute later and then reconnected:

```
2019-09-08 18:35:22,713 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2019-09-08 18:35:27,010 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2019-09-08 18:35:27,038 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2019-09-08 18:35:27,043 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2019-09-08 18:35:27,050 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-09-08 18:35:27,272 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
[...]
2019-09-08 18:35:30,365 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2019-09-08 18:35:30,383 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-09-08 18:35:30,432 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.6 (Github)"
2019-09-08 18:35:30,479 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
```

The reason for the disconnect *might* be visible in the generated `serial.log`s (once that's enabled)

seltix5, 2019-09-11:
For this test I deleted all logs, then I "connect" octoprint to the printer and then "reboot system", those are the logs (the printer was NOT shutdown) :
[octoprint.log](https://github.com/foosel/OctoPrint/files/3601815/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/3601816/serial.log)

seltix5, 2019-09-11:
wait, i'm wrong, the printer did shutdown because of the PSU control relay. It probably is the case where the port is not yet registered.
In this case shouldn't it try again 1 minute or 2 later?

seltix5, 2019-09-11:
yet if I set the correct port instead of AUTO it does connect without problem at boot...
[octoprint.log](https://github.com/foosel/OctoPrint/files/3601864/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/3601865/serial.log)

foosel, 2019-10-31:
> In this case shouldn't it try again 1 minute or 2 later? because when the printer is power up at the same time with octoprint it will reproduce the same problem...

Maybe it should, but that's not how it is implemented right now. It expects the printer to be available on startup, if the printer isn't available then it can't connect. In any case, if it connects when you select the port explicitly but not when set to AUTO I don't think it's the port not yet being registered.

I just pushed the above patch to the `maintenance` branch that contains a potential fix (I have the suspicion that for some reason the port is set to `None` instead of the value `AUTO`, which would make the autoconnect routine not trigger so far) and also adds a bunch of logging around the autoconnect code. Please check that out, run it and see if it fixes your problem.

seltix5, 2019-10-31:
Thanks.
How do I test it?

foosel, 2019-10-31:
Switch your installation to the `maintenance` branch as described here:

https://community.octoprint.org/t/how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image/206

foosel, 2019-11-05:
@seltix5 any results yet?

seltix5, 2019-11-05:
hello, sorry for the delay, I did not had time yet to try it. I already create a backup of my pi just in case and I will change the branch today and try it for the next days

seltix5, 2019-11-08:
hello,
I was going to try the maintenance branch but before that I try to reproduce the problem just to be sure (I was not runing AUTO since my last testing) and i'm unable to reproduce the problem anymore, i'm on OctoPrint 1.3.12, I dont understand. Did that version include a related update?

foosel, 2019-11-11:
> Did that version include a related update?

Hm... no. It shouldn't.

seltix5, 2019-11-12:
i dont understand then :S i set the connection back to auto and I did not had the problem again

foosel, 2019-11-12:
Oh! I understand though. At least possibly. The issue I fixed above on `maintenance` could be caused by a value in the `port` setting that should be interpreted as "AUTO" but wasn't properly. It might have gotten in there at some point, then you set it to a fixed port, and setting it back to "AUTO" wrote the correct value.

seltix5, 2019-11-12:
but in the previous tests I did test the manual port to confirm it worked as expected and set the auto again and the problem did not disapear.... :octocat:

foosel, 2019-11-12:
Hm. Then I don't know either. The code that I fixed was faulty though, so let's just hope that was it.

seltix5, 2019-11-12:
I supose we can close this issue then and if i found the same problem again in the future i will reply here

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel