OctoPrint/OctoPrint #3274: OctoPrint cannot start (TypeError: unsupported operand type(s) for *: 'EPollIOLoop' and 'float')

Lissanro, 2019-09-17:
#### What were you doing?
I have installed OctoPrint in Orange Pi Lite:

1. sudo apt-get install git python-pip python-dev python-wheel python-setuptools psmisc dh-python python-stdeb fakeroot python-all _# Install some packages necessary to compile and install OctoPrint_
2. git clone https://github.com/foosel/OctoPrint.git; cd OctoPrint
3. sudo python setup.py --command-packages=stdeb.command bdist_deb && sudo dpkg -i --force-depends deb_dist/*.deb _# Created .deb package with OctoPrint and install it_
4. sudo apt -f install _# Install OctoPrint dependencies_
5. sudo pip install flask_assets flask_login==0.2.11 awesome_slugify cachelib chainmap emoji pylru sarge sentry_sdk==0.7.7 _# When .deb package of OctoPrint was building, not all dependencies where found in Armbian repository so install them with pip_
6. octoprint serve _# Attempt to run OctoPrint_

#### What did you expect to happen?

OctoPrint start without errors.

#### What happened instead?

OctoPrint crashed with uncaught exception:
```
Traceback (most recent call last):
  File "/usr/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.11', 'console_scripts', 'octoprint')()
  File "/usr/lib/python2.7/dist-packages/octoprint/__init__.py", line 609, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/octoprint/cli/server.py", line 206, in serve_command
    ignore_blacklist)
  File "/usr/lib/python2.7/dist-packages/octoprint/cli/server.py", line 118, in run_server
    octoprint_server.run()
  File "/usr/lib/python2.7/dist-packages/octoprint/server/__init__.py", line 525, in run
    user_settings=dict(websocket_allow_origin="*" if enable_cors else ""))
  File "/usr/lib/python2.7/dist-packages/octoprint/vendor/sockjs/tornado/router.py", line 112, in __init__
    self._sessions_cleanup.start()
  File "/usr/lib/python2.7/dist-packages/tornado/ioloop.py", line 1209, in start
    self._schedule_next()
  File "/usr/lib/python2.7/dist-packages/tornado/ioloop.py", line 1237, in _schedule_next
    self._update_next(self.io_loop.time())
  File "/usr/lib/python2.7/dist-packages/tornado/ioloop.py", line 1244, in _update_next
    callback_time_sec *= 1 + (self.jitter * (random.random() - 0.5))
TypeError: unsupported operand type(s) for *: 'EPollIOLoop' and 'float'
```

#### Did the same happen when running OctoPrint in safe mode?

Yes, "octoprint serve --safe" produces the same error message.

#### Version of OctoPrint

% octoprint --version
octoprint, version 1.3.11

#### Operating System running OctoPrint

Linux. I have used Etcher to burn Armbian Buster minimal 5.3 image, then I created octoprint account and added it to sudoers.

#### Printer model & used firmware incl. version

Since I cannot run OctoPrint, my printer model does not matter.

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://pastebin.com/pnsjQp1B

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

foosel, 2019-09-17:
That's an unsupported way of installing it. If you need to get it to work I suggest going the recommended `virtualenv` route. 

I sadly lack the resources to look into getting this approach supported and this will stay the case for the foreseeable future so I'm closing this.

If you want to look into it yourself, I'll be happy to get a related pull request however!

closed 2019-09-17 by foosel

Lissanro, 2019-09-18:
Thank you for quick reply and pointing me to the solution. I did not know that generating .deb could cause issues like that. In case somebody finds this issue with search engine in the future, I'm going to mention how I solved this issue. It turned out that getting it to work (after uninstalling broken .deb package) is as simple as running "sudo pip install octoprint". Manually installing with "pip install" octoprint dependencies was unnecessary since they are installed automatically by pip.

foosel, 2019-09-18:
In order to avoid issues with system wide installed dependencies however, instead of `sudo pip install octoprint` you instead should rather go for something like

```
virtualenv /opt/octoprint
/opt/octoprint/bin/pip install octoprint
```

Also do *not* run OctoPrint under user root but add a dedicated user for it.