OctoPrint/OctoPrint #3275: [RC Feedback] Feedback on the 1.3.12rc2 Release Candidate

foosel, 2019-09-17:
Please provide *general* feedback on your experience with the 1.3.12rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

b-morgan, 2019-09-17:
Updated from 1.3.12rc1 to 1.3.12rc2 (on OctoPi 16.0) successfully.

gryzlov, 2019-09-18:
Trying to print ~30 mb file. Gcode viewer freeze page after agreement with warning until page reload

foosel, 2019-09-18:
@gryzlov please share the file

gryzlov, 2019-09-18:
[Sample_Spool_fixed.zip](https://github.com/foosel/OctoPrint/files/3625245/Sample_Spool_fixed.zip)

foosel, 2019-09-18:
@gryzlov Thank you, I appear to have a reproduction. Looks like a bug fix I did increased memory consumption way too much, I'll have to find a different solution.

schnello, 2019-09-19:
Update from RC1 to RC2 without problems. Some smaller prints (2-6 hours) are fine.

kazibole, 2019-09-19:
A few SD card prints, no problems!

ejjenkins, 2019-09-20:
After updating to RC2 the WebCam Tab plugin no longer works and the webcam is back on the control tab.
OctoPrint:  Version 1.3.12rc2
OctoPi:  Version 0.16.0, running on Raspberry Pi 3 Model B Rev 1.2

foosel, 2019-09-21:
@ejjenkins Can you please file a full bug report? Also - is this a regression compared to RC1 or 1.3.11 stable?

ejjenkins, 2019-09-21:
I will file a bug report.  It worked properly with RC1.

foosel, 2019-09-23:
1.3.12rc3 is out, new ticket here: https://github.com/foosel/OctoPrint/issues/3283

closed 2019-09-23 by foosel