OctoPrint/OctoPrint #3276: USB Support

Lino77, 2019-09-17:
Marlin supports since the latest USB stick from 2.0 bugfix, unfortunately I do not see the data on Octopi is it possible to integrate them

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

AmiSMB, 2019-09-18:
Octoprint supports SD card via gcode commands M20 through to M30 but it is dependent on the board you are using as well as the commit point of Marlin bugfix-2.0.x you are using as well as which OS you are using this on. This information would help alot.

foosel, 2019-09-18:
As already mentioned, from OctoPrint's side there is really nothing to do here. If the firmware simply exposes the data on the USB stick the same way it already does it for SD cards, it will just work out of the box.

closed 2019-09-18 by foosel