OctoPrint/OctoPrint #3278: Error when enabling/disabling/installing/uninstalling plugins while printer is paused

mdelecate, 2019-09-18:
#### What were you doing?
1. Paused an ongoing print from the web interface
2. Clicked on OctoPrint Settings 
3. Clicked on Plugin Manager
4. Attempted to enable/disable/install/uninstall plugins 

#### What did you expect to happen?
Expected the action on the plugin to complete, e.g. enable/disable/install/uninstall plugin. 

#### What happened instead?
An error in red pops up: "Something went wrong / Please consult octoprint.log for details"

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.12rc2

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Artillery/Evnovo Sidewinder X1
Marlin 1.1.9

#### Browser and version of browser, operating system running browser
Max OS 10.14.6 (18G87) 
Safari Version 12.1.2 (14607.3.9)

#### Link to octoprint.log
https://gist.github.com/mdelecate/05b3c1ee499b8211a5181780c7ff7c5c#file-octoprint-log

#### Link to contents of terminal tab or serial.log
https://gist.github.com/mdelecate/05b3c1ee499b8211a5181780c7ff7c5c#file-serial-log

#### Link to contents of Javascript console in the browser
[image]

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2019-09-18:
Denying plugin management while a print job is active (whether printing or paused is irrelevant here) is correct. But it should inform about that in the web interface and not even offer the options. That it does is a bug.

mdelecate, 2019-09-18:
Holy moly you're as fast as lightning!!!

foosel, 2019-09-18:
Coincidence ;) Trying to be as productive as possible while sitting in a waiting room.

foosel, 2019-09-18:
Fixed by the above commit, already on `maintenance`, soon `devel`, to be released with 1.3.**13** (1.3.12 is already frozen).

mdelecate, 2019-09-18:
All from a waiting room!!! 🤯

commit 04528a63 (2019-09-18): 🐛 PMGR: Disable plugin mgmt while printer is busy

Not just when only actively printing.

Closes #3278
closed 2020-03-04 by foosel