OctoPrint/OctoPrint #3281: 1.3.12rc2 - WebCam Tab is not working after installing release candidate 2

ejjenkins, 2019-09-21:
After updating to 1.3.12rc2 the WebCam Tab plugin stopped working.  The tab is still there, but it is blank and the webcam is displayed on the control tab.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

https://pastebin.com/dsv1zdcQ

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

ejjenkins, 2019-09-21:
[image]
[image]

ejjenkins, 2019-09-21:
It seems most of my comments disappeared after posting...

same behavior on Firefox and Chrome
OctoPrint 1.3.12RC2
OctoPi 0.16.0

foosel, 2019-09-21:
I'll also need

  * Browser and version of browser, operating system running browser
  * Link to contents of Javascript console in the browser

ejjenkins, 2019-09-21:
My camera settings seem to have all been reset.  I will put the settings back and restart after the print finishes and post the results.

ejjenkins, 2019-09-22:
Google Chrome:  Version 77.0.3865.90 (Official Build) (64-bit)
(Latest version of FireFox as well, but I am not on that system to check version number)

Windows 10 Home 1903 build 18362.356

OctoPrint-WebcamTab:  https://github.com/malnvenshorn/OctoPrint-WebcamTab

https://pastebin.com/c6PBrJiJ

ejjenkins, 2019-09-22:
All the camera settings are correct (I have a 2nd printer with the same camera and it works fine...it is not running the rc2.  I removed and reinstalled the WebCam Tab plugin...no change.

foosel, 2019-09-23:
I cannot reproduce this. I just installed the plugin on 1.3.12rc2 and it works just like it should.

[image]

[image]

[image]

foosel, 2019-09-23:
Also not reproducible on a fresh 1.3.12rc3

ejjenkins, 2019-09-23:
When I get home from work this evening I will remove plugins, one at a time to see if I can narrow this down.

ejjenkins, 2019-09-23:
I removed the OctoPrint-Dashboard plugin and the WebCam tab worked perfectly.  Not sure what the issue was or how it interacted, but it works now...I uninstalled both updated to RC3...reinstalled both and WebCam Tab is still broken with the OctoPrint-Dashboard plugin installed.  So it is not a bug with the RC.

https://github.com/StefanCohen/OctoPrint-Dashboard

ejjenkins, 2019-09-23:
One thing to mention, they both worked prior to installing the RC2.

foosel, 2019-09-24:
If you also updated the dashboard plugin when you updated to RC2 I'm pretty sure what's going on is that the webcam support it added in 1.2.0 released right around the same time is what caused the issue.

As I said, there's nothing that changed between rc1 and 2 that even touches the webcam integration, so this looks like the more likely culprit.

Closing this since it's third party plugins interfering with each other.

closed 2019-09-24 by foosel

foosel, 2019-09-24:
FWIW, I just reproduced under 1.3.11 with a combo of both plugins in current versions. Downgrading to an older version of the dashboard plugin without webcam support, 1.1.0 specifically, makes the issue go away.

ejjenkins, 2019-09-24:
That makes perfect sense...I did update the dashboard plugin at about the same time...thank you for following up with this.