OctoPrint/OctoPrint #3283: [RC Feedback] Feedback on the 1.3.12rc3 Release Candidate

foosel, 2019-09-23:
Please provide *general* feedback on your experience with the 1.3.12rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

  * ❓ [#3281 - WebCam Tab plugin (third party) is not working after installing 1.3.12rc2](https://github.com/foosel/OctoPrint/issues/3281) - works fine here

### Unreproduced and information for further analysis missing

b-morgan, 2019-09-23:
Updated from 1.3.12rc2 to 1.3.12rc3 (on OctoPi 16.0) successfully.

JohnOCFII, 2019-09-24:
I recently wiped my OctoPi 0.13.0 and put down a fresh copy of OctoPi 0.16.0.  (I made a backup, but forgot to upload it to my desktop -- doh!)

So - Clean OctoPi with just OctoSlack 1.9.1, Telegram Notifications 1.5.0 and M73 ETA Override 1.0.1 plugins on OctoPrint 1.3.12rc3 is working well.  I tested: GCode Viewer, Terminal, Time-lapse, uploading and starting a print.  This is connected to a Prusa MK3S with MMU2S.  One of the prints I did was launch a single material MMU2S print.  That also worked fine.

schnello, 2019-09-25:
Update from RC2 to RC3 done, server is online and all (incl some prints) works fine.

NovaViper, 2019-09-28:
My printer seems to be stopping very often now. It'll just stop randomly (everything stops including the extruder motor) and sit for a few seconds before moving again. This has started since RC3

foosel, 2019-10-01:
Sounds very good all in all so far. Currently fighting off the flu, once I've succeeded in that I'll push out 1.3.12, unless something comes up in the meantime.

rknobbe, 2019-10-01:
The Cura Octoprint Connection plugin (on the Cura side) seems to not work anymore. Not sure if the problem is on the Cura side or the Octoprint side.  I get an "octoprint is busy, unable to start a new job" error from Cura.  I'll keep researching.

foosel, 2019-10-03:
> The Cura Octoprint Connection plugin (on the Cura side) seems to not work anymore.

Can you be more specific? Also pinging @fieldOfView just in case...

fieldOfView, 2019-10-03:
I likely won’t have time to look in to this until monday.

fieldOfView, 2019-10-07:
@rknobbe, I cannot reproduce this issue. What version of Cura, and what version of the plugin are you using?

gcnix, 2019-10-08:
> > The Cura Octoprint Connection plugin (on the Cura side) seems to not work anymore.
> 
> Can you be more specific, e.g. share logs? Also pinging @fieldOfView just in case...

I have the same issue with 1.3.11. I found that once I connect Octoprint with printer ("verbinden"), it takes a few seconds for Cura to detect the printer being available. I then see an offer in Cura to queue the job, which I decline and with the next attempt it works. I believe it is only a timing issue.

arhi, 2019-10-09:
works flawlessly for me, printed bunch of stuff on rc3 on 2 printers (one smoothieware based, other marlin based) no issues what so ever

louispires, 2019-10-11:
This version was working great until this morning when I had my first ever error between Firmware and OctoPrint: https://github.com/foosel/OctoPrint/issues/3303

foosel, 2019-10-22:
1.3.12 has been released

closed 2019-10-22 by foosel