OctoPrint/OctoPrint #3285: [Request] Display ETA alongside/below "Print Time Left"

webdes03, 2019-09-23:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
During a print, you often check the "Print Time Left" to estimate the completion time of a print. As opposed to doing the math separately, it would be useful (particularly for long running jobs) to have the estimated completion time displayed alongside (or under) the "Print Time Left" metric in the State panel. In the event that the completion date is tomorrow (or on another future date), use of a +1 (+n) label could be used to indicate how many days, avoiding the need for screen real estate for a full date.

Estimated Completion: 13:21
Estimated Completion: 07:48 +1 (use of the +1 indicator to indicate next day)

**Describe alternatives you've considered**

**Additional context**

ManuelMcLure, 2019-09-29:
[OctoPrint-Display-ETA](https://plugins.octoprint.org/plugins/octoprint-eta/) is a plugin that does what you want.

foosel, 2019-10-03:
As mentioned there is a plugin that does this already.

closed 2019-10-03 by foosel