OctoPrint/OctoPrint #3288: [Request] Filterable List in Plugin Manager

jneilliii, 2019-09-26:
**Is your feature request related to a problem? Please describe.**
Now that the plugin list is a long list rather than paged, It would be beneficial to have a text filtering option for minimizing the list when trying to find something to enable/disable/uninstall.

**Describe the solution you'd like**
A simple text box at the top of the list, similar to the list when installing a plugin, that filters what's there based on the entered text.

jneilliii, 2019-09-26:
I will look at this capability in the near future and potentially submit a PR for it.

foosel, 2019-09-26:
Needed a quick win

[image]

jneilliii, 2019-09-28:
Thanks Gina. I see a couple of other filter buttons up there as well.  Any possibility to get enabled and disabled added to those?

foosel, 2019-10-08:
I'll take a look

foosel, 2019-10-25:
[image]

jneilliii, 2019-10-25:
Your awesome @foosel, thanks.

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel