OctoPrint/OctoPrint #3289: WebCam does not display as expected

ProtomakerSprint, 2019-09-28:
WebCam does not display as expected.

**_Hope this helps
[image]
[image]
[OctoPrint 1.3.12rc3 no webcam shown.txt](https://github.com/foosel/OctoPrint/files/3665654/OctoPrint.1.3.12rc3.no.webcam.shown.txt)

[serial.log](https://github.com/foosel/OctoPrint/files/3665655/serial.log)
[image]
[image]
[image]

_**

**Control Tab** displays WebCam for about two seconds only, shows Keyboard actions but I believe these do not function

**WebCam Tab** displays blank

Only  **/webcam/?action=stream** displays stream of the WebCam

Attached Images and some Logs

jneilliii, 2019-09-28:
This is an issue with the dashboard plugin and webcam tab plugin both being installed. Check out the dashboard plugin's issues...

foosel, 2019-09-29:
Yep, known third party plugin issue with on relation to the rc.

closed 2019-09-29 by foosel